contiki-os/contiki #1582: Issue when using ADC on CC26xx

chenek, 2016-04-05:
I use the following code in hello-world.c and run it on my CC2650 LaunchPad (or CC2650STK get the same issue) and I see my ADC sample code can only read ADC 6 times. Then, it just stucks in  AUXADCReadFifo. Does anyone know what’s wrong?

#include "contiki.h"
#include "ti-lib.h"
#include "driverlib/aux_adc.h"
#include "driverlib/aux_wuc.h"

#include <stdio.h> /* For printf() */

//#define CLOCK_SECOND 128
#define SECONDS 2

char buf[32];

/*---------------------------------------------------------------------------*/
PROCESS(hello_world_process, "Hello world process");
AUTOSTART_PROCESSES(&hello_world_process);
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(hello_world_process, ev, data)
{
  PROCESS_BEGIN();
    unsigned short i;
	uint16_t singleSample;

    printf("ADC demo\r\n");
    while(1)
    {       
		//intialisation of ADC
		ti_lib_aon_wuc_aux_wakeup_event(AONWUC_AUX_WAKEUP);
		while(!(ti_lib_aon_wuc_power_status_get() & AONWUC_AUX_POWER_ON))
		{ }

		// Enable clock for ADC digital and analog interface (not currently enabled in driver)
		// Enable clocks
		ti_lib_aux_wuc_clock_enable(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK | AUX_WUC_SMPH_CLOCK);
		while(ti_lib_aux_wuc_clock_status(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK | AUX_WUC_SMPH_CLOCK) != AUX_WUC_CLOCK_READY)
		{ }

		printf("clock selected\r\n");
		
		// Connect AUX IO7 (DIO23, but also DP2 on XDS110) as analog input.
		AUXADCSelectInput(ADC_COMPB_IN_AUXIO7);  
		printf("input selected\r\n");
		
		// Set up ADC range
		// AUXADC_REF_FIXED = nominally 4.3 V
		AUXADCEnableSync(AUXADC_REF_FIXED,  AUXADC_SAMPLE_TIME_2P7_US, AUXADC_TRIGGER_MANUAL);
		printf("init adc --- OK\r\n");

        //Trigger ADC converting
        AUXADCGenManualTrigger();
        printf("trigger --- OK\r\n");
        
        //reading adc value
        singleSample = AUXADCReadFifo();

        printf("%d mv on ADC\r\n",singleSample);
        
        //This timer crashes the ADC's FIFO (please let me know why) 
                //Please use the clock_delay_usec as shown below
        //etimer_set(&timer, CLOCK_SECOND);
        //PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&timer));
		//shut the adc down
		//AUXADCDisable();
		//printf("disable --- OK\r\n");

        for(i=0;i<10;i++)  
            clock_delay_usec(10000);  //can't go up to 65535 that's why i use a for loop
    }

  PROCESS_END();
}
/*---------------------------------------------------------------------------*/

lligios, 2016-04-05:
Is watchdog enabled? 
take a look to main.c process.. check if whatchdod is enabled and "refreshed"... 
you are looping inside this process without giving any chance to reset the watchdog countdown..

Furthermore pay attention to deep sleep.. when cc2650 goes in sleep mode (during yeald or waiting for timers) AON it is powered off .. not everything goes up well ... 
Try to compile with no sleep mode at all..

chenek, 2016-04-05:
I try to refresh watchdog but it still get stuck at the same place. Any other idea?

        singleSample = AUXADCReadFifo();

        printf("%d mv on ADC\r\n",singleSample);
        watchdog_periodic();

lligios, 2016-04-05:
did you tried to change LPM mode?

take a look at row 56

#define LPM_MODE_MAX_SUPPORTED LPM_MODE_DEEP_SLEEP

in cpu/cc26xx-cc13xx/lpm.h

g-oikonomou, 2016-04-05:
I've updated the format on the OP to enclose the snippet within a code block.

Whenever your process yields, the AUX will be powered off. Coming out of deep sleep will not automatically turn AUX back on, so you have to re-initialise it.

We needed ADC functionality ourselves so @atiselsts has made some extensions to the LPM logic that allows ADC usage without having to worry about how/when to fire up AUX.

So:

> //This timer crashes the ADC's FIFO (please let me know why) 

It's actually not the timer, it's the yield in the following line.

I would put `AUXADCDisable();` before the `PROCESS_WAIT_EVENT_UNTIL(...);`

I cannot immediately understand why you need this for loop with the `clock_delay_usec()`.

You should also be careful about using dynamic variables in process threads. As a rule of thumb: Don't do it unless you understand exactly how they will behave.

chenek, 2016-04-05:
@lligios I have tried to set "#define LPM_MODE_MAX_SUPPORTED LPM_MODE_SLEEP" but I still see the problem.
@g-oikonomou I use clock_delay_usec to have some delay between two ADC reading. I also try to put "AUXADCDisable" right after ADC reading with AUXADCReadFifo but I still see the problem. What do you mean to be careful about using dynamic variables in process threads?

marco-pistone, 2016-04-05:
try this snippet before the line AUXADCGenManualTrigger();

if ((AONWUCPowerStatusGet() & AONWUC_AUX_POWER_DOWN))
	{
                printf("Try to wakeup\n");
		AONWUCAuxWakeupEvent(AONWUC_AUX_WAKEUP);
		while (!(AONWUCPowerStatusGet() & AONWUC_AUX_POWER_ON))
		{
		}
	}

chenek, 2016-04-05:
I add those code and test again. However, it still get stuck. Any further suggestion?

#include "contiki.h"
#include "ti-lib.h"
#include "driverlib/aux_adc.h"
#include "driverlib/aux_wuc.h"
#include "dev/watchdog.h"

#include <stdio.h> /* For printf() */

//#define CLOCK_SECOND 128
#define SECONDS 2

char buf[32];

/*---------------------------------------------------------------------------*/
PROCESS(hello_world_process, "Hello world process");
AUTOSTART_PROCESSES(&hello_world_process);
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(hello_world_process, ev, data)
{
  PROCESS_BEGIN();
    unsigned short i;
	uint16_t singleSample;

    printf("ADC demo\r\n");
    while(1)
    {       
		//intialisation of ADC
		ti_lib_aon_wuc_aux_wakeup_event(AONWUC_AUX_WAKEUP);
		while(!(ti_lib_aon_wuc_power_status_get() & AONWUC_AUX_POWER_ON))
		{ }

		// Enable clock for ADC digital and analog interface (not currently enabled in driver)
		// Enable clocks
		ti_lib_aux_wuc_clock_enable(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK | AUX_WUC_SMPH_CLOCK);
		while(ti_lib_aux_wuc_clock_status(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK | AUX_WUC_SMPH_CLOCK) != AUX_WUC_CLOCK_READY)
		{ }

		printf("clock selected\r\n");
		
		// Connect AUX IO7 (DIO23, but also DP2 on XDS110) as analog input.
		AUXADCSelectInput(ADC_COMPB_IN_AUXIO7);  
		printf("input selected\r\n");
		
		// Set up ADC range
		// AUXADC_REF_FIXED = nominally 4.3 V
		AUXADCEnableSync(AUXADC_REF_FIXED,  AUXADC_SAMPLE_TIME_2P7_US, AUXADC_TRIGGER_MANUAL);
		printf("init adc --- OK\r\n");

		if ((AONWUCPowerStatusGet() & AONWUC_AUX_POWER_DOWN))
		{
			printf("Try to wakeup\n");
			AONWUCAuxWakeupEvent(AONWUC_AUX_WAKEUP);
			while (!(AONWUCPowerStatusGet() & AONWUC_AUX_POWER_ON))
			{}
		}
        //Trigger ADC converting
        AUXADCGenManualTrigger();
        printf("trigger --- OK\r\n");
        
        //reading adc value
        singleSample = AUXADCReadFifo();

        printf("%d mv on ADC\r\n",singleSample);
        watchdog_periodic();
        //This timer crashes the ADC's FIFO (please let me know why) 
                //Please use the clock_delay_usec as shown below
        //etimer_set(&timer, CLOCK_SECOND);
        //PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&timer));
		//shut the adc down
		AUXADCDisable();
		printf("disable --- OK\r\n");

        for(i=0;i<10;i++)  
            clock_delay_usec(10000);  //can't go up to 65535 that's why i use a for loop
    }

  PROCESS_END();
}
/*---------------------------------------------------------------------------*/

marco-pistone, 2016-04-05:
sorry,
put hte code not before the gentrigger but before the readfifo

chenek, 2016-04-05:
It still doesn't work. I don't think it would work because I already reinitialize ADC in while loop.

marco-pistone, 2016-04-05:
i developed an ADC driver, at the beginning i had similar problems solved now, but i use a ctimer to read ADC values not a loop.
in the callback function i reinitialize the ADC in the same manner, with little differences.

chenek, 2016-04-05:
Can you share your code example?

marco-pistone, 2016-04-05:
This is the code tha read the ADC values, the ctimer callback call this function that returns the adc value.

static int
crs_value(int type)
{
	uint32_t singleSample;
	uint32_t voltage;

	if ((AONWUCPowerStatusGet() & AONWUC_AUX_POWER_DOWN))
	{
		PRINTF_CRS(WARNING"  CRS Try Wakeup"RESET"\r\n");
		AONWUCAuxWakeupEvent(AONWUC_AUX_WAKEUP);
		while (!(AONWUCPowerStatusGet() & AONWUC_AUX_POWER_ON))
		{
		}
	}

	AUXWUCClockEnable(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK);
	PRINTF_CRS(GOOD"  CRS AUXWUCClockEnable"RESET"\r\n");

	AUXADCSelectInput(ADC_COMPB_IN_AUXIO3);
	PRINTF_CRS(GOOD"  CRS AUXADCSelectInput"RESET"\r\n");

	// Set up ADC range	
	AUXADCEnableSync(AUXADC_REF_FIXED, AUXADC_SAMPLE_TIME_2P7_US, AUXADC_TRIGGER_MANUAL);
	PRINTF_CRS(GOOD"  CRS AUXADCEnableSync"RESET"\r\n");

	AUXADCGenManualTrigger();
	PRINTF_CRS(GOOD"  CRS Manual Trigger"RESET"\r\n");

	if ((AONWUCPowerStatusGet() & AONWUC_AUX_POWER_DOWN))
	{
		PRINTF_CRS(WARNING"\tCRS Try Wakeup 2"RESET"\r\n");
		AONWUCAuxWakeupEvent(AONWUC_AUX_WAKEUP);
		while (!(AONWUCPowerStatusGet() & AONWUC_AUX_POWER_ON))
		{
		}

		AUXWUCClockEnable(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK);
		PRINTF_CRS(WARNING"\tCRS AUXWUCClockEnable 2"RESET"\r\n");

		AUXADCSelectInput(ADC_COMPB_IN_AUXIO3);
		PRINTF_CRS(WARNING"\tCRS AUXADCSelectInput 2"RESET"\r\n");

		// Set up ADC range	
		AUXADCEnableSync(AUXADC_REF_FIXED, AUXADC_SAMPLE_TIME_2P7_US, AUXADC_TRIGGER_MANUAL);
		PRINTF_CRS(WARNING"\tCRS AUXADCEnableSync 2"RESET"\r\n");
		
		AUXADCGenManualTrigger();
		PRINTF_CRS(WARNING"\tCRS Manual Trigger 2"RESET"\r\n");
	}

	if ((AONWUCPowerStatusGet() & AONWUC_AUX_POWER_DOWN))
	{
		printf(BOLD""CRITICAL"\tCRS Try WakeUp AUX 3"RESET"\r\n");
		AONWUCAuxWakeupEvent(AONWUC_AUX_WAKEUP);
		while (!(AONWUCPowerStatusGet() & AONWUC_AUX_POWER_ON))
		{
		}
	}

	singleSample = AUXADCReadFifo();
	PRINTF_CRS(INFO"  CRS %"PRIu32" mv on ADC"RESET"\r\n", singleSample);
	
	PRINTF_CRS(GOOD"  CRS AUXDisable "RESET"\r\n");
	AUXADCDisable();

	switch (type) {
	case CURRENT_SENSOR_VALUE_RAW:
		return singleSample;
		break;
	case CURRENT_SENSOR_VALUE_VOLTAGE:
		voltage =voltage_value(singleSample);
		return voltage;
		break;
	}

	return singleSample;
}

marco-pistone, 2016-04-05:
pay attention the PRINTF_CRS MACRO doesn't work for you is a macro tha print out colored output in the serial

chenek, 2016-04-06:
I revise my ADC code to use etimer and it works fined now. I think @g-oikonomou is correct that I cannot use a while-loop to do this ADC reading.

closed 2016-04-06 by chenek

msolters, 2016-06-10:
Hi @chenek , what do you mean when you say you revised the code to use `etimer`?  Do you think you could share an example of that, which you know to work?  I am also having a very hard time getting `AUXADCReadFifo()` and `AUXADCDisable()` to not freeze (curiously, in code that has always worked until I pulled in the latest Contiki changes)

chenek, 2016-06-11:
You can refer to http://sunmaysky.blogspot.tw/2016/04/contiki-cc2650-adc-demo-using-dio23-as.html?m=1

g-oikonomou, 2016-06-11:
Check out the new stuff from #1684

msolters, 2016-06-13:
Wow! George, that looks great. Thank you!