contiki-os/contiki #1584: error "Reading outside memory" in hmac_sha1(prf_data)‏

htm1112, 2016-04-06:
I am trying to test some methods (mainly session resumption) to IKEv.2 inside the Ipsec protocol, using Contiki OS as implemented by the Cooja simulation environment. 

I need your help to assist me incorporate the session resumption protocol code in C, so that I can resume my simulation.

I tried my best to develop the code, but every time I try to run it, it always gives me the same error " reading out of memory"

I trace the error in IPsec and IKEv.2 protocol code that I download it from this like https://github.com/vjutvik/Contiki-IPsec 

 and found the error appear when call hmac_sha1(prf_data); function. the error message is  "Reading outside memory: 0xfa8b6".

the file path of this function is: contiki-ipsec\core\net\ipsec\ike\prf.c

/**

  * PRF as defined in the RFC

  */

void prf(sa_prf_transform_type_t prf_type, prf_data_t *prf_data)

{

  switch (prf_type) {

    case SA_PRF_HMAC_SHA1:         // MUST

    hmac_sha1(prf_data);

    break;

    

    case SA_PRF_AES128_CBC:      // SHOULD+

    PRINTF(IPSEC "Error: Not implemented\n");

    break;

    

    default:

    PRINTF(IPSEC "Error: Unknown PRF request\n");

  }  

}

-------------------------------------------------------------

the code function  hmac_sha1(prf_data); 

the file path of this function is: contiki-ipsec\core\lib\hmac-sha1\hmac-sha1.c

/**

 * Performs HMAC-SHA1-96. Call this function from your program.

 *

 * @param out     Output from HMAC-SHA1-96 (must be at least 96/8=12 bytes)

 * @param key     The key to use

 * @param keylen  The length of the key

 * @param data    The data to hash

 * @param datalen The length of the data

 */

//void hmac_sha1_96( uint16_t datalen, uint8_t * out, uint8_t * key, uint8_t keylen, uint8_t * data )

void hmac_sha1( hmac_data_t *hmac_data )

{

  uint8_t * out=hmac_data->out;

  uint8_t * key= (uint8_t *) hmac_data->key;

  uint8_t keylen=hmac_data->keylen;

  uint8_t * data=hmac_data->data;

  uint16_t datalen=hmac_data->datalen;

  dy_printf( "Data after %d %d %d %d %d\n", (int)out, (int)key,(int)keylen,(int)data, (int)datalen );

	// Variables

	uint8_t padded_key[ SHA1_B ];

	uint8_t padded_key_xor_ipad[ SHA1_B ];

	uint8_t sentence_a[ SHA1_B + datalen ];

	uint8_t hash_output_a[ SHA1_L ];

	uint8_t padded_key_xor_opad[ SHA1_B ];

	uint8_t sentence_b[ SHA1_B + SHA1_L ];

	uint8_t hash_output_b[ SHA1_L ];

	init();

	// Hash overlength keys

	if( keylen > SHA1_B ){

		hash_sha1( padded_key, key, keylen ); // Assumes SHA1_L <= SHA_B

		create_padded_key( padded_key, padded_key, SHA1_L ); // Adds the zeros

	}

	else{

		create_padded_key( padded_key, key, keylen );

	}

	dy_printf("datelen:%d\n",(int)datalen);

	PRINT_BUF("padded_key",padded_key,SHA1_B);

	xor_array( padded_key_xor_ipad, padded_key, ipad, SHA1_B );

	PRINT_BUF("padded_key_xor_ipad",padded_key_xor_ipad,SHA1_B);

	merge_arrays( sentence_a, padded_key_xor_ipad, SHA1_B, data, datalen );

	PRINT_BUF("sentence_a",sentence_a,SHA1_B+datalen);

	hash_sha1( hash_output_a, sentence_a, SHA1_B + datalen );

	PRINT_BUF("hash_output_a",hash_output_a,SHA1_L);

	xor_array( padded_key_xor_opad, padded_key, opad, SHA1_B );

	PRINT_BUF("padded_key_xor_opad",padded_key_xor_opad,SHA1_B);

	merge_arrays( sentence_b, padded_key_xor_opad, SHA1_B, hash_output_a, SHA1_L );

  hash_sha1(hmac_data->out, sentence_b, SHA1_B + SHA1_L );

	// Old truncate code

	//hash_sha1( hash_output_b, sentence_b, SHA1_B + SHA1_L );

	//truncate( out, hash_output_b, 96/8 );

}

-----------------------------

I use Ubuntu 14.04 LTS 32-bit

 

gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4

Copyright (C) 2013 Free Software Foundation, Inc.

 

msp430-gcc (GCC) 4.7.2 20120920 (mspgcc dev 20120911)

Copyright (C) 2012 Free Software Foundation, Inc.

 

I using Contiki-IPsec 2.7 OS, also I’m using Wismote in Cooja simulation

 

Any idea to overcome this issue is highly appreciated. 

Thanking you in advanced.

Haytham Qalalwi

alignan, 2016-04-06:
This is a branch out of Contiki's main repository, you could either create an issue there, or just wait for feedback in Contiki mailing list (where you have already opened a thread).

closed 2016-04-06 by alignan

htm1112, 2016-04-06:
sorry about that, but I relay need help and no one answer me, and my thesis depend on this, if you kindly give me an answer or any one how can.
thanks and sorry again for inconvenience.