contiki-os/contiki #1586: Travis failing due to arm-gcc version no longer being available

g-oikonomou, 2016-04-06:
Since #1504 we are explicitly specifying what version of the toolchain to use for the travis arm job. Unfortunately, this version is no longer available on the PPA, apt fails to install it and the test falls over.

We can obviously revert the change from #1504 but then we will go back to the previous situation of travis using a different toolchain version than the one we specify in various READMEs. We have already agreed that we don't want that.

I've not 100% made up my mind how we want to fix this. @bthebaudeau @alignan any thoughts?

g-oikonomou, 2016-04-06:
Thinking aloud: One option is to forget about apt altogether and wget the desired linux tarball from launchpad.net.

alignan, 2016-04-06:
I was going to suggest the same.

bthebaudeau, 2016-04-06:
:+1: for `wget`.

PR #1587 (2016-04-06): Fix failure of ARM-AAPCS travis job
Up till now we have been installing arm-gcc using apt from the team-gcc-arm-embedded ppa.

As discussed in #1453 and implemented in #1504, we have decided to lock travis to use the specific version of the toolchain that is documented in the READMEs of relevant platforms. However, the PPA no longer hosts this version, apt fails to install arm-gcc and as a result the job fails too.

This pull changes the travis job to wget an installation tarball for the desired version, instead of trying to install via apt.

Fixes #1586
closed 2016-04-06 by bthebaudeau