contiki-os/contiki #1591: CC26xx Delayed Etimer Servicing after Sleep

bkozak-scanimetrics, 2016-04-11:
On CC26xx and CC13xx, we schedule an RTC interrupt to occur at the next etimer. When we wake up, it is possible that we woke due to some event other than the RTC. If this is the case, then we will not have another timer tick occur until the interrupt which was scheduled for the last etimer.

This will only be a problem if, after waking up, we schedule an etimer which would expire before the current value of ```etimer_next_expiration_time()```. 

To fix this we can set an RTC interrupt to occur at the next tick interval when we wake up. Another solution might be to modify the etimer scheduling code so that it can schedule an interrupt to occur on its own (thus eliminating the need for timer tick interrupts altogether).

g-oikonomou, 2016-07-16:
@bkozak-scanimetrics Billy is this one still open after recent LPM-related merges?

bkozak-scanimetrics, 2016-07-18:
The problem still seems to be there. 

I don't see any code which will schedule the next timer tick after waking.

atiselsts, 2016-07-20:
This is needed and the recent changes certainly did not add any code for handling this case (I was under impression that it's already done, but did not check).

Should not be too difficult to fix, but I personally won't have time to do that in the short term. This is done correctly on JN516x, where the clock is completely tickless now - going this route would require more substantial changes. See `AppWarmStart()` in `contiki-jn516x-main.c` and the code in file `clock.c` for details. A simpler option is to leave the ticking on, but reschedule if immediately after wakeup.