contiki-os/contiki #1593: Commit 26Mar 2016 breaks communications with 6LBR

tiagodiogo, 2016-04-13:
hello,
I've been using Contiki-OS together with the border router 6LBR from the cetic repo on my master thesis.
Today i updated my contiki-os repo and after i pulled to the latest version i stopped having plain text communications with my border router.
after trying to find out what went wrong i could isolate the problem to the commit made on the 26Mar 2016 (actually a lot of commits) but i could narrow it to:

plaintext communication working on 3b9fa19
plaintext communication not working on ab4a0e0

the problem is that this was a huge contribution from @joakimeriksson with lots of changes and i don't know what can be the cause of the lack of communication.

in the border router log file i still see wireless inputs, but then the node never gets added to the topology not it appears on the border router online sensors.

i know this is a terrible request, but do you have any idea what can be originating the problem and how to fix it?

thank you,
Tiago

chenek, 2016-04-14:
Do you check the discussion in #1573?

tiagodiogo, 2016-04-15:
i haven't seen that discussion, but my 6lbr prefix is already fd00 so that is not the issue

joakimeriksson, 2016-04-19:
The issue might be that 6LBR make use of aaaa:: as compression prefix 0 and Contiki after the fix make use of fd00:: as the compression prefix 0. Try fd01:: as prefix and see if that works.

g-oikonomou, 2016-07-16:
I am testing with latest master + PR #1779 (which I will merge when travis passes).

I am running a recent-ish version of 6lbr (Version : 1.4.0 (Contiki-6lbr-1.4.0)) on BBB, with a CC2531 USB dongle slip-radio, using prefix `fd00::/64` and 6LoPWAN Context 0 `fd00::`.

Web demo works perfectly fine on SmartRF06+CC2650EM. I am running mosquitto on my OS X, device connected and publishing just fine. I can now also access the device from my mac, web pages and CoAP all work and the board is advertising itself over BLE. I am not seeing any problems whatsoever.

I am closing this.

closed 2016-07-16 by g-oikonomou