contiki-os/contiki #1607: Wiki pages in non-english?

joakimeriksson, 2016-04-20:
I just saw that there is some new Wiki pages for Contiki. These seems to be in Chinese and are for me very hard to read. Is there any specific new policy for the Wiki?

joakimeriksson, 2016-04-24:
Do you want me to draft a wiki-contribution policy?

alignan, 2016-04-24:
Normally I would prefer to have contribution being consistent, but then again I prefer to have people contributing in an useful way, perhaps for Chinese users this documentation is more useful.

What is the specific problem? are there English pages being edited in Chinese, or new pages in Chinese?

joakimeriksson, 2016-04-24:
Take a look at some of the pages far down the list such as:
https://github.com/contiki-os/contiki/wiki/回调定时器（Callback-timer）