contiki-os/contiki #1614: OpenMote/contiki

gardasg, 2016-04-21:
I git latest OpenMote/contiki and tried to build rpl-udp and webserver-ipv6 examples for openmote-cc2538 platform. I've got the following error related to sicslowpan context:

In file included from ../../../core/./contiki.h:38:0,
                 from ../../../core/net/ipv6/sicslowpan.c:63:
../../../core/net/ipv6/sicslowpan.c: In function 'sicslowpan_init':
../../../platform/openmote-cc2538/./contiki-conf.h:528:32: error: 'UIP_DS6_DEFAULT_PREFIX_0' undeclared (first use in this function)
   addr_contexts[0].prefix[0] = UIP_DS6_DEFAULT_PREFIX_0; \
                                ^
../../../core/net/ipv6/sicslowpan.c:1758:3: note: in expansion of macro 'SICSLOWPAN_CONF_ADDR_CONTEXT_0'
   SICSLOWPAN_CONF_ADDR_CONTEXT_0;
   ^
../../../platform/openmote-cc2538/./contiki-conf.h:528:32: note: each undeclared identifier is reported only once for each function it appears in
   addr_contexts[0].prefix[0] = UIP_DS6_DEFAULT_PREFIX_0; \
                                ^
../../../core/net/ipv6/sicslowpan.c:1758:3: note: in expansion of macro 'SICSLOWPAN_CONF_ADDR_CONTEXT_0'
   SICSLOWPAN_CONF_ADDR_CONTEXT_0;
   ^
../../../platform/openmote-cc2538/./contiki-conf.h:529:32: error: 'UIP_DS6_DEFAULT_PREFIX_1' undeclared (first use in this function)
   addr_contexts[0].prefix[1] = UIP_DS6_DEFAULT_PREFIX_1; \
                                ^
../../../core/net/ipv6/sicslowpan.c:1758:3: note: in expansion of macro 'SICSLOWPAN_CONF_ADDR_CONTEXT_0'
   SICSLOWPAN_CONF_ADDR_CONTEXT_0;
   ^
make: *** [obj_openmote-cc2538/sicslowpan.o] Error 1

Note: the build is working for cc2538dk platform!

alignan, 2016-04-21:
Openmote has been merged into Contiki now, are you using the latest Contiki's master, or an OpenMote branch? it is not clear.  CC: @peretuset

peretuset, 2016-04-21:
I just compiled the ipv6/rpl-udp with the latest Contiki HEAD and it compiled successfully.
@gardasg What repository did you use? Since the OpenMote platform is already supported in Contiki you no longer need to use the OpenMote/contiki branch.
Can you please retry?

gardasg, 2016-04-21:
Thank you! I used the OpenMote branch. Now it's working fine.

closed 2016-04-21 by gardasg