contiki-os/contiki #1615: A number of problems with CC26xx CCA

g-oikonomou, 2016-04-21:
We currently use:

```
  /* Configure CCA settings */
  cmd->ccaOpt.ccaEnEnergy = 1;
  cmd->ccaOpt.ccaEnCorr = 1;
  cmd->ccaOpt.ccaEnSync = 0;
  cmd->ccaOpt.ccaCorrOp = 1;
  cmd->ccaOpt.ccaSyncOp = 1;
```

So, we are basically relying on observed RSSI and the number of correlation peaks observed during the last 8-symbol period. For the channel to be busy, we require both of those signals to be busy. SFD reception is ignored altogether. This has a number of problems:

* We are not standards-compliant. The standard (and common sense) demands the channel to be always reported as busy when an SFD is found.
* It is possible to receive a frame correctly and in its entirety, but `receiving_packet()` will be signaling 'false' through the entire packet reception. Thanks to @atiselsts for reporting this in the first place, which is what led to discovering this bug

In order to be standards-compliant, we need to start by setting:
```
  cmd->ccaOpt.ccaEnSync = 1;
  cmd->ccaOpt.ccaSyncOp = 0;
```

We then need to modify `receiving_packet()` to ignore RSSI and Correlator and to simply inspect the value of `ccaInfo.ccaSync` returned by `CMD_IEEE_CCA_REQ`.

Lastly, we need to choose a CCA mode among the 3 specified in .15.4. If a sync has _not_ been found, then:

1. Based on RSSI value only
2. Based on correlator only
3a. Based on both of the above, using an OR logic: Busy if either the RSSI and Correlator test signal busy, clear otherwise.
3b. Based on both of the above, using an AND logic: Busy if both the RSSI and Correlator test signal busy, clear otherwise.

bkozak-scanimetrics, 2016-04-21:
> It is possible to receive a frame correctly and in its entirety, but receiving_packet() will be signaling 'false' throughout the entire packet reception. Thanks to @atiselsts for reporting this in the first place, which is what led to discovering this bug

So if I understand correctly, ```receiving_packet()``` will only return a false negative if a packet with a low RSSI is received?

g-oikonomou, 2016-04-21:
That's right. If RSSI level is below `IEEE_MODE_RSSI_THRESHOLD` (currently -90) then `receiving_packet()` will return false negatives. We observed this only when nodes were to far apart from each-other.

g-oikonomou, 2016-04-21:
But if signal is strong enough, `receiving_packet()` will return true, provided of course that the correlator also signals a busy channel (which is what should happen when we detect a sync).

g-oikonomou, 2016-04-22:
Some more info here.

If sync signals "idle", Atis and I are tempted to change our logic to signal channel busy if either the RSSI signal is "busy" OR the correlator signal is "busy", instead of using the current "AND" logic.

Furthermore, the "sync" signal goes idle at the end of frame reception, but it appears this happens a few usecs before the RX queue gets marked as "finished". This means we may end up in a situation where the RF driver's `read_frame()` may get called before the queue gets marked as "finished", in which case `read_frame()` will fail to read our the packet thinking there isn't one. This is super problematic with TSCH.

Our current workaround is to change `read_frame()` to block shortly until the RX queue entry gets marked as either "finished" or "free" (the latter is what will happen if the frame gets dropped e.g. due to a CRC error)

bkozak-scanimetrics, 2016-04-22:
> If sync signals "idle", Atis and I are tempted to change our logic to signal channel busy if either the RSSI signal is "busy" OR the correlator signal is "busy", instead of using the current "AND" logic.

I remember trying this out at some point just after the CC2650 platform was added. The performance was significantly worse because of a high number of false positives. Maybe with all of the development that has happened since then this is no longer a problem but I think you should watch out for this issue when you test this change.

g-oikonomou, 2016-04-22:
Well it's a bit of a tradeoff really, I think. Either you get false "channel clear" signals when there is in fact traffic and you therefore cause collisions, or you erroneously suppress a TX so as to not cause a collision, when in reality you could have just sent

bkozak-scanimetrics, 2016-04-22:
Definitely a trade-off.

In addition to false positive collisions, however, I also remember that it was causing ContikiMAC to drop packets somehow (I didn't and still don't understand how this might happen). It now seems to me that this may have been buggy behaviour; which is why I say that maybe this is no longer a problem.

Still, just something to watch-out for.

PR #1709 (2016-06-15): TSCH port for CC26xx  
This pull request adds a full-featured (HW timetamps, polling mode, etc.) TSCH port on CC26xx. This port has been tested for several months in a small testbed with custom CC26xx SoC based devices at the University of Bristol.

Depends on #1708 and #1684 (#1708 is used as the base branch, and some cherry-picked commits included from #1684).

Fixes #1615.
closed 2016-06-15 by g-oikonomou