contiki-os/contiki #1623: Can not use malloc on CC26XX/CC13XX platform

xukai871105, 2016-04-24:
I just write a very simple program to test malloc function on CC1310 platform.
`#include "contiki.h"

#include <stdio.h> /* For printf() */
#include <stdlib.h>
/*---------------------------------------------------------------------------*/
PROCESS(hello_world_process, "Hello world process");
AUTOSTART_PROCESSES(&hello_world_process);
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(hello_world_process, ev, data)
{
  PROCESS_BEGIN();

  char *mem = (char *)malloc(20);
  sprintf(mem, "Hello %s", "World");
  printf("%s\n", mem);
  free(mem);
  
  PROCESS_END();
}`

A error happen and I could not solve it.
"/usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/lib/armv7-m/libc.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `**_sbrk**'
collect2: error: ld returned 1 exit status"

I just modify the makefile.cc26xx-cc13xx file .
Add "MODULES += lib/newlib" 
But it has no effect.

I also test the same program on the cc2538 platform. It can work well.

Finally, please forgive my poor english.

tonnenpinguin, 2016-04-24:
Have a look at [this](https://github.com/contiki-os/contiki/wiki/Memory-allocation) wiki page that describes contikis approach to dynamic memory allocation!

closed 2016-05-14 by g-oikonomou

aignacio, 2016-09-01:
Hello @xukai871105 

Did you got a way to use malloc on **CC2650**? I try to implement the sbrk but still does not work...