contiki-os/contiki #1627: Contiki Regression Tests

joakimeriksson, 2016-04-24:
I really do like that Contiki have regression test so that we always know that things work after adding new code or changing things. But they are significantly undocumented. I would suggest one of the following two approaches:

* README.md page per subset of tests describing each test with its idea (what it tests), intended limitations (e.g. too few nodes in the routing table) and how the script evaluates the success (and if there is something specific with the code).

or

* A wiki page with description of the tests (same info as above)

The reason for this is to make it easier for people to contribute code and understand why a specific test fails. It is today a bit of a pain to dig in to the tests and try to figure out what it actually does.

simonduq, 2016-04-25:
Yes that would be extremely welcome!

joakimeriksson, 2016-04-25:
I will do an attempt at the RPL tests first then - and submit a PR on that to get it started!