contiki-os/contiki #1641: Unable to get powertrace results in CC2650 Sensortag 2.0

drako65, 2016-04-30:
I tried running powertrace on Sensortag 2.0 CC2650, but all the values are zero. 
Did I miss something in the code?
In the Makefile I added APPS+=powertrace
in the code I added #include powertrace.h
and powertrace_print
Would could be wrong?
Thanks in advance for your help.

Ramil

g-oikonomou, 2016-05-18:
https://github.com/contiki-os/contiki/wiki#Help

closed 2016-05-18 by g-oikonomou

Ashokrlz, 2017-05-27:
Hi I'm also facing the same issue., all the values are zero when running powertrace example in cc2650 launchpad. The link that you've provided doesn't seem to have a valid answer for this. 

Regards

beleyj, 2018-03-07:
Powertrace uses an application called Energest so be sure you have energest running! To add energest to a program:
The Energest modules is enabled by configuring ENERGEST_CONF_ON to 1. This is easiest done by adding a project-conf.h file to the product build with the following content.

#define ENERGEST_CONF_ON 1

Source: https://github.com/contiki-ng/contiki-ng/wiki/Documentation:-Energest