contiki-os/contiki #1649: Can contiki mote get real-world time?

wufucious, 2016-05-03:
In my project, all sensors need to know real-world time, say "dd/mm/year, 12:12", when they produce and store data. Any suggestion? Examples? 

From my pointof view, all Contiki time library is usefulness because none of them are designed to get real-world time value.

alignan, 2016-05-03:
We have this for the `RE-Mote` using an on-board RTCC:
https://github.com/contiki-os/contiki/tree/master/examples/zolertia/zoul/rtcc

Other alternative would be to poll the date to a NTP server.

You are welcome to contribute.

marco-pistone, 2016-05-03:
i had implemented the conversion of unix time polled from my border router, It is not accurate to the second, but it is sufficient to my needs

vsaw, 2016-05-03:
FYI: (sort of) Duplicate of #1648

g-oikonomou, 2016-05-18:
I consider this to have been answered. Closing

closed 2016-05-18 by g-oikonomou