contiki-os/contiki #1650: IoT / 6LoWPAN TODO List - suggestion

joakimeriksson, 2016-05-04:
This is a refresh of the TODOs for Contiki's 6LoWPAN stack - a more short-term TODO List than a full road-map with a focus on IoT and 6LoWPAN.

Lots of things have happened recently with the 6LoWPAN/RPL stack but we still have important steps todo: (unordered bullets):

- Make the RPL implementation even more feature complete (e.g. get in the non-storing PR and several other RPL related PRs)
- Make some final fixes on the 6LoWPAN layer to make it fully compliant with the RFC
- Improve / clean-up interface between Contiki and the stack (now we are using lots of different timers, lots of APIs that are Contiki only - people are now and then using the stack as a separate thing - see Zephyr project).
- Add CSL (802.15.4e) so that we have both CSL and TSCH for Contiki
- Add support for 802.15.4 active scan to detect networks (PANs) automatically (CMD frames and Beacon frames)
- Add support for sleepy devices (today the 6LoWPAN stack more or less assumes non-sleepy behaviour).
- Improve the LWM2M engine - today it is very much a non-complete LWM2M implementation
- Add support for DTLS - at least a clear description on how to use TinyDTLS or similar
- Enable support for multiple radios - e.g. how to handle 2.4 and subGHz in the same device - not necessarily multiple IP-interfaces but some information in the neighbor tables are needed
- Add specific regression tests for scalability (DAO ACK, etc) and new features (I will add LWM2M regression test).
- Document existing regression tests for easier understanding when things go red.

I am mainly working with tuning and improving storing mode for RPL but also working quite a lot with sleepy devices and getting longer battery lifetime (>10 years). Since Yanzi is happy to contribute quite a lot of their improvements to Contiki and I would be happy to put efforts into the upstreaming of these features but I would also like to see commitment from others too. Not just adding requirements to the list but actually putting in hours to get there. BTW: increased merge team would also be a nice thing - to increase speed of PR evaluation. I am happy to evaluate stuff that I know - like RPL PRs, timers, etc - and I am trying to do that as much as I can. But the merge team also needs people doing that (otherwise it is a wasted effort for me).

YinxuWang, 2016-05-16:
I like this suggestion. 
I was working on Enable support for multiple radios by adding more entries in the neighbor and routing tables.

CogPy, 2016-11-23:
A nice feature would be a function in the REST interface that would query whether it's okay to go to sleep, i.e., all packets have been sent.