contiki-os/contiki #1655: Is it possible to test er-rest-example using CC2650DK or CC2650STK?

chenek, 2016-05-05:
I tried to build er-rest-example for CC2650DK but I get the following link errors. Does anyone know how to fix it?

$ make TARGET=srf06-cc26xx BOARD=srf06/cc26xx
"C:/Program Files (x86)/Git/bin/find.exe": paths must precede expression
Usage: "C:/Program Files (x86)/Git/bin/find.exe" [path...] [expression]
"C:/Program Files (x86)/Git/bin/find.exe": paths must precede expression
Usage: "C:/Program Files (x86)/Git/bin/find.exe" [path...] [expression]
  CC        ../../cpu/cc26xx-cc13xx/./ieee-addr.c
  AR        contiki-srf06-cc26xx.a
  CC        ../../cpu/cc26xx-cc13xx/./fault-handlers.c
  CC        ../../cpu/cc26xx-cc13xx/lib/cc26xxware/startup_files/ccfg.c
  CC        ../../cpu/cc26xx-cc13xx/lib/cc26xxware/startup_files/startup_gcc.c
  CC        er-example-server.c
  LD        er-example-server.elf
er-example-server.co: In function `process_thread_er_example_server':
er-example-server.c:(.text.process_thread_er_example_server+0x74): undefined reference to `res_hello'
er-example-server.c:(.text.process_thread_er_example_server+0x7c): undefined reference to `res_push'
er-example-server.c:(.text.process_thread_er_example_server+0x84): undefined reference to `res_toggle'
er-example-server.c:(.text.process_thread_er_example_server+0x94): undefined reference to `res_event'
er-example-server.c:(.text.process_thread_er_example_server+0x98): undefined reference to `res_separate'
collect2.exe: error: ld returned 1 exit status
../../cpu/cc26xx-cc13xx/Makefile.cc26xx-cc13xx:98: recipe for target 'er-example-server.elf' failed
make: *** [er-example-server.elf] Error 1
rm er-example-server.co obj_srf06-cc26xx/ccfg.o obj_srf06-cc26xx/fault-handlers.o obj_srf06-cc26xx/startup_gcc.o

g-oikonomou, 2016-05-05:
Successfully built on OS X just now. Could it be a Cygwin thing? Or could it something as simple as having to make clean first?

chenek, 2016-05-06:
I build with InstantContiki on VMPlayer and it works. It's weird that it cannot be built by Cygwin. Do you have any idea?

oliverschmidt, 2016-05-06:
Hi,

If this would be Windows I'd guess that your PATH is messed up and that
therefore the "wrong" 'find' is called. No idea if this might apply to OSX
too...

Regards, Oliver

chenek, 2016-05-06:
But I can build cc26xx-web-demo without problem.

greg-king5, 2016-05-07:
Both Oliver and George are correct.  That listing shows at least two bugs.

1. That `find.exe` program is for Windows (the Cygwin version would be `/usr/bin/find`).  I think that you haven't installed Cygwin's "find utilities" package.  Run Cygwin's Setup application.  Go to the "Select Packages" page.  Type **findu** into the Search box.  Expand the "Base" category.  Install the `findutils` package.  After that, Cygwin's program loader will see Cygwin's version of `find` before it sees that Windows' version.
2. The listing shows that only a few files were compiled, at that time.  You need to remove old object files before building for a (newly chosen) target and board:
```sh
$ make clean TARGET=srf06-cc26xx BOARD=srf06/cc26xx
```

greg-king5, 2016-05-07:
I just realized that step number two in my previous message probably isn't needed by you.  You probably attempted that make command several times; and, the other files were built -- properly -- by the previous runs.  You showed to us only the very last attempt.  If I am right, then the "find utilities" bug is the only thing that needs to be fixed.

chenek, 2016-05-07:
Thanks a lot. It works now.

closed 2016-05-07 by alignan