contiki-os/contiki #1658: predictable mac-level performance in low power wireless under interference lab in contiki

opealek, 2016-05-10:
Ref paper : predictable mac-level performance in low power wireless under interference
Can someone guide me through how to setup the experimental setup for this paper. I tried but I am new to contikimac. SO I need someone to just put me through what to do. This should be a simulation in my case

Extarct for the lab is : 
Our testbed consists of a sender-receiver couple and an interferer mote that we place in an office environment close to each other (< 1m). All threes nodes are Zolertia Z1 plat- forms that feature a CC2420 radio and run Contiki 2.7. Each experiment lasts for 30 minutes during which we transmit around 900 packets. We use IEEE 802.15.4 channel 26 since it does not overlap with WiFi [27]. The packet size is 127 bytes, which is the maximum transmission unit in IEEE 802.15.4. We have chosen this packet size as it represents the worst case since long packets are more likely to be corrupted than short ones. 
To create interference, we use the CC2420’s test mode to generate a continuous carrier [6]. We create semi-periodic interference as described by Boano et al. [10] using a two- state process that consists of a clear channel state and an in- terference state.

nvt, 2016-05-12:
Although I'm a co-author of the cited paper, I will not provide a comment here because it is the wrong forum. The issues here should be related to the development of Contiki. Furthermore, it is highly unclear what your request ("I need someone to just put me through what to do") entails.

closed 2016-05-12 by nvt

opealek, 2016-05-14:
Sorry, can you direct me to the right forum ... Please its my ignorace

opealek, 2016-05-14:
Sorry it is my ignorance. Can you direct me to the right forum. I greatly accept my failure here and accept the correction made.
Can you direct me to the right forum/..

nvt, 2016-05-16:
No problem. The usual channel for such matters is to send e-mail directly to a paper author.