contiki-os/contiki #1659: CC1310 is 10db down compared to TI SmartRF Studio 7

jimmycmlo, 2016-05-10:
In reference to an issue reported on sourceforge:

https://sourceforge.net/p/contiki/mailman/contiki-developers/thread/CAFBGMVOe5SPfnBr%2BrRMNAdkp3Hm%2BtmCFysv6AzixYs%2BYzHLR3g%40mail.gmail.com/#msg34873321

I was able to reproduce the same issue.

I added a debug print to read_frame() in prop_mode.c in cpu/cc26xx-cc13xx/rf-core

	int8_t rssi = (int8_t)data_ptr[len];
  	printf("read_frame len=%d, rssi=%d\n", len, rssi);

which print out the rssi of each received packet.

The output is some packets have higher RSSI than other, especially after a transmit

read_frame len=65, rssi=-52
read_frame len=65, rssi=-41
read_frame len=65, rssi=-52
read_frame len=65, rssi=-52
read_frame len=95, rssi=-41
read_frame len=65, rssi=-53

This is a real RF problem, not just an RSSI reporting error. I was able to confirm that with a range test.

This is also a receiver problem, not a transmitter problem. I was able to confirm that using the SmartRF 7 Studio in continuous receive mode. The reported RSSI is the higher one.

I was able to narrow it down to radio duty cycling. So after the radio is turn off and back on, the initial reception seems to have lower RSSI. If I disable radio duty cycling by either using the nullrdc_driver or just return immediately in the off() function without turning off the radio, all packets will have the higher rssi.

It looks like the power up sequence did not fully wakeup the radio until later. 

I tried putting delays in the power up sequence without success.

I reviewed the power up sequence in rf_core_power_up() in rf-core.c, but cannot find anything obvious.

I hope you can shed some light on this issue.

g-oikonomou, 2016-05-12:
That's very interesting, I wonder if it's related with overrides. Could it also be linked to #1348?

jimmycmlo, 2016-05-12:
I don't think it is related to #1348.

This RSSI is appended to the packet by the radio when it was received. As I mentioned before, this is real RF issue, not just a RSSI reporting issue. The radio is seeing a weaker signal but I was able to confirm the transmitted signal is fine. It looks like the radio is not setup correctly when it was turn on.

The only time I see a proper RSSI is when it was immediately after a transmit operation. So one giant hack is to put the radio in transmit mode first, abort it immediately and then put it back into receive mode. It works but very ugly, not to mention increased power consumption. It may also have other unintended consequences. I am hopping for a more elegant solution.

I have asked TI to review the radio power up and down sequence but I have not gotten a reply yet.

jonnteolsson, 2016-06-05:
We will make a round of updating the cc13xxware/cc26xxware and look into the prop-mode RF driver in June timeframe. Expect some PRs.

arurke, 2016-08-17:
@jimmycmlo I'm curious if you have been able to test this after #1756 ? My tests showed no difference but would be great with a second input.

g-oikonomou, 2016-08-17:
My gut feeling is that we will see little change if at all. My gut feeling points to the direction of "us" (Contiki) and "them" (Smart RF Studio) using different overrides and/or RF patches.

Reports from testing very welcome! We need to track this one down.

emreerisgen, 2016-09-02:
We have our own custom design CC1310 boards and we have the same low db problem. I want to share our results with you.

We made the measurements with 2 boards which have 2 meter distance between. We use SMA 868 mhz 2db antennas. And also we have R&S FSV Signal and Spectrum Analyzer. We used SmartRF Studio to get the RSSI level and also Contiki Here are our results.

1. We used 1 of our boards on TX mode - The other one connected to the R&S FSV Signal and Spectrum Analyzer with cable. ----   RSSI value from Spectrum Analyzer = -33db

2. We used 1 of our boards on TX mode - The other one on RX mode and read the RSSI value from TI SmartRF Studio ---- RSSI value = -43db

3. We flashed one of our boards with cc26xx-web-demo - The other one is connected to the 6lbr as slip radio ---- RSSI value which read from Contiki = -53db

All test was made at the same location no change. 

At short with : Spectrum Analyzer = -33db
                       SmartRF Studio       = -43db
                       Contiki                     = -53db

jonnteolsson, 2016-09-29:
I think, by only looking at the code, that I have spotted an error in the IEEE802.15.4g compliant settings we are using for CC1310 radio driver. The  preamble is only 3 bytes (in smartrf-settings.c .preamConf.nPreamBytes = 0x3) while the IEEE802.15.4g specification mandates 4 bytes. My theorry at the moment is that this shorter preamble leads to that the RSSI reported by the radio core is wrong. I will try with  .preamConf.nPreamBytes = 0x4 and let you know the result.

chenek, 2016-10-03:
I tried to change .preamConf.nPreamBytes from 0x3 to 0x4 and I cannot access to sensor web page anymore.

jonnteolsson, 2016-10-03:
Hi YK,

Did you make the same change also on the Edge Router side? Does the node connect to the network at all?

Texas Instruments Norway AS, Postboks 264 Skøyen, Hoffsveien 70 C, NO-0213 Oslo, Norway. Org. NO 980499480 MVA

chenek, 2016-10-03:
Hi jonnas,
Do you mean to rebuild 6lbr or slip radio? I rebuild slip radio but not 6lbr.

jonnteolsson, 2016-10-03:
Slip radio should be enough. Did the node join the network at all?

chenek, 2016-10-03:
I can see Sensor show up on Sensor tab of 6lbr wen page but when I try to open Sensor web page, I cannot access it. Do you test it and see it works?

jonnteolsson, 2016-10-03:
Not been able to test yet.

tj57, 2016-10-04:
I would like to confirm problem.
I have changed  .preamConf.nPreamBytes from 0x3 to 0x4 on server and node. In log I see "New node created ..." but I can't ping node, my ping statistics is 504/506 (99%) lost.

g-oikonomou, 2016-10-13:
Hello @tj57, I'm surprised you actually managed to get 2 packets through with `nPreamBytes` = 4. If you change this, you also have to adjust the sync word on the transmitter and receiver.

Try setting the following, I think it should work but untested:

In `smartrf_settings_cmd_prop_tx_adv`:
`.syncWord = 0x5555904e`

In `smartrf_settings_cmd_prop_rx_adv`
`syncWord0 = 0x5555904e`

tj57, 2016-10-17:
I have changed syncWords and I see the same problem, only one ping works.
```
 DEBUG: NODE: New node created 2001:db8:2::212:4b00:a29:486a

--- 2001:db8:2::212:4b00:a29:486a ping statistics ---
1043 packets transmitted, 1 received, +451 errors, 99% packet loss, time 1044184ms
rtt min/avg/max/mdev = 14785.714/14785.714/14785.714/0.000 ms, pipe 15
```

Johan-Henry, 2016-10-24:
I can confirm the above issue with our custom cc1310 boards using rpl-border-router and some of our nodes sending UDP packets. From testing it is clear that the issue only exist when using contikimac. When the receiver is kept on with NETSTACK_MAC.off(1) in the same code, the issue is not present and the RSSI is 10db higher compared to using contikimac. I have tried the above suggestions to change .preamConf.nPreamBytes from 0x3 to 0x4 and tx/rx syncWord = 0x5555904e, and these setting alterations makes no difference to the RSSI issue and in fact only worsen the throughput. 

It would be great if a solution could be found in the near future. Thanks for all the efforts here

g-oikonomou, 2016-10-24:
@Johan-Henry thanks for the report, this makes me wonder whether this is related to @fros4943 comments here: https://github.com/contiki-os/contiki/pull/1348#issuecomment-158434603

Johan-Henry, 2016-10-24:
I doubt it, I just tried to implement a delay as mentioned by @fros4943 but it did not make a difference.

g-oikonomou, 2016-11-06:
SmartRF Studio code export includes this override:

```
    // Rx: Set RSSI offset to adjust reported RSSI by +5 dB
    (uint32_t)0x00FB88A3,
```

g-oikonomou, 2016-11-13:
To increase preamble length, we were forgetting `.formatConf.nSwBits`.

So, here is the full changeset required for `smartrf-settings.c` on both sides. I've tested it at 863MHz. It's kinda slow but it does work. I do sincerely doubt this will make any difference in terms of the RSSI though.

```
diff --git a/cpu/cc26xx-cc13xx/rf-core/smartrf-settings.c b/cpu/cc26xx-cc13xx/rf-core/smartrf-settings.c
index 24cef30..8773d0f 100644
--- a/cpu/cc26xx-cc13xx/rf-core/smartrf-settings.c
+++ b/cpu/cc26xx-cc13xx/rf-core/smartrf-settings.c
@@ -81,9 +81,9 @@ rfc_CMD_PROP_RADIO_DIV_SETUP_t smartrf_settings_cmd_prop_radio_div_setup =
   .symbolRate.preScale = 0xf,
   .symbolRate.rateWord = 0x8000,
   .rxBw = 0x24,
-  .preamConf.nPreamBytes = 0x3,
+  .preamConf.nPreamBytes = 0x4,
   .preamConf.preamMode = 0x0,
-  .formatConf.nSwBits = 0x18,
+  .formatConf.nSwBits = 0x20,
   .formatConf.bBitReversal = 0x0,
   .formatConf.bMsbFirst = 0x1,
   .formatConf.fecMode = 0x0,
@@ -150,7 +150,7 @@ rfc_CMD_PROP_TX_ADV_t smartrf_settings_cmd_prop_tx_adv =
   .preTrigger.triggerNo = 0x0,
   .preTrigger.pastTrig = 0x1,
   .preTime = 0x00000000,
-  .syncWord = 0x0055904e,
+  .syncWord = 0x5555904e,
   .pPkt = 0,
 };
 /*---------------------------------------------------------------------------*/
@@ -182,7 +182,7 @@ rfc_CMD_PROP_RX_ADV_t smartrf_settings_cmd_prop_rx_adv =
   .rxConf.bAppendRssi = 0x1,
   .rxConf.bAppendTimestamp = 0x0,
   .rxConf.bAppendStatus = 0x1,
-  .syncWord0 = 0x0055904e,
+  .syncWord0 = 0x5555904e,
   .syncWord1 = 0x00000000,
   .maxPktLen = 0x0000, /* To be populated by the driver. */
   .hdrConf.numHdrBits = 0x10, /* 16: .4g mode */
```

g-oikonomou, 2016-11-13:
Folks, can you please edit `smartrf-settings.c` and change the `overrides[]` array. Add the following two lines somewhere _before_ `(uint32_t)0xFFFFFFFF,`. Let me know if you see increased reported RSSIs.

```
    HW_REG_OVERRIDE(0x6084, 0x35F1),
    (uint32_t)0x00F688A3,
```

@jimmycmlo @arurke @emreerisgen @chenek @Johan-Henry

Johan-Henry, 2016-11-14:
Thanks @g-oikonomou , here are my initial test results using one rpl-border-router and one node with about 2m distance between them on my desk:
Using nullrdc without override: RSSI -22, communication seems stable
Using nullrdc with override: RSSI -18, communication seems stable
Using contikimac without override: RSSI -32, communication seems stable
Using contikimac with override: Cannot get stable communication, seem to get a hand full of packets through after a reboot of the border router and the node, but thereafter rf-comms completely dies.

g-oikonomou, 2016-11-14:
Thanks that's very useful. What device are you using?

Johan-Henry, 2016-11-14:
CC1310F128RGZR

I have also tested these for long distance comms in the business park where I work, and by using nullrdc I get relatively close in distance to where the sock TI dev nodes get in high speed mode, however when using contikimac I dont even get a quarter of the distance.

g-oikonomou, 2016-11-14:
Can I ask you to try using each of those two overrides individually.

My gut feeling is that the ContikiMAC distance thing may be related to the moving average discussion in #1348

Thanks for providing info!

Johan-Henry, 2016-11-14:
Sorry, did not say that the above mentioned SOC is on a custom board.

So when using nullrdc and only HW_REG_OVERRIDE(0x6084, 0x35F1) in same setup as mentioned above, I get a RSSI of about -23 and only using (uint32_t)0x00F688A3 I get a RSSI of about -18. Unfortunately I am not having much luck getting stable comms with either individually when using contikimac. Contikimac does however work well when both are not used.

g-oikonomou, 2016-11-14:
Hmm I do find this somewhat strange. The `HW_REG_OVERRIDE` is meant to freeze RSSI on sync found, this I guess could impact behaviour. The other one though is only meant to adjust what the radio is reporting to us, so I cannot immediately understand why it would mess around with connectivity.

I did see some funny results whereby two theoretically identical devices (CC1350 STKs) were behaving differently using the same set of overrides: One was fine and the other was unacceptably slow. Still digging.

Thanks again for the reports, very helpful. I'll do some of my own tests and see whether I can reproduce what you are reporting.

arurke, 2016-11-27:
@g-oikonomou You need more tests of the HW_REG_OVERRIDE?

g-oikonomou, 2016-11-27:
I've not had a chance to look at this in a few days, but anything that might help understand patches and overrides will certainly help. I am starting to suspect this may have to do with how we are duty cycling the radio very frequently with ContikiMAC - Could be very wrong though.

g-oikonomou, 2016-12-04:
@arurke @Johan-Henry Andreas, Johan I've had a chat with TI and I have pushed a new update under #1932 and this includes new RF patches and overrides. Reported RSSIs seem a lot more sane now and also those overrides discussed above work in my setup.

Can you sanity-check please?

Cheers

arurke, 2016-12-04:
Nice work @g-oikonomou! I see 8-10 dB increase on RSSI on received frames with #1932. Sanity-wise it also looks ok from my POV. Tested on cc1310em+smartrf (Don't have any cc1350).

I also now noticed that before #1932 the RSSI of received ACKs are ~8-10 dB higher than received 6lowpan frames. I believe this confirms @jimmycmlo comment https://github.com/contiki-os/contiki/issues/1659#issue-154085481 on "The only time I see a proper RSSI is when it was immediately after a transmit operation"

Are you now able to tell if this is/was a problem with the reported RSSI or actual reduced signal strength? @jimmycmlo how did you measure range? Could it be that you just met the PROP_MODE_RSSI_THRESHOLD too early due to low reported RSSI?

g-oikonomou, 2016-12-04:
Thanks for testing

I think it has to do with how the radio averages things out in order to report a value. I would suspect the RSSIs seen from ACK frames may related to the fact that the radio has been on for a while. Is this no longer the case with #1932?

Naturally if the radio reports very low signal, we'll be dropping frames that we might have been able to receive, hence range would decrease.

arurke, 2016-12-04:
In #1932 ACK and data frames have same RSSI.

g-oikonomou, 2016-12-04:
Fantastic news, I had forgotten Jimmy had made that observation tbh.

Johan-Henry, 2016-12-05:
I have briefly tested with #1932 now and results seems much better on my platform. I will conduct some more tests and report back if I see anything strange regarding this issue. Thank you for the great work.

jonnteolsson, 2016-12-05:
Confirm I see stable RSSI readings after applying #1932

arurke, 2017-07-01:
Increasing the preamble to 4 bytes to comply with .4g was discussed here, but in the end the PR did not include this change. Any rationale behind leaving it as is?

jonnteolsson, 2017-07-03:
Yes, the comment from me was wrong. :) We are already using 4 bytes preamble.

arurke, 2017-07-03:
Are you sure?
```
 .preamConf.nPreamBytes = 0x3,
```
https://github.com/contiki-os/contiki/blob/master/cpu/cc26xx-cc13xx/rf-core/smartrf-settings.c#L182

jonnteolsson, 2017-07-05:
Yes, positive! :)

We are using one byte 0x55 in the sync word as a little trick here.

https://github.com/contiki-os/contiki/blob/master/cpu/cc26xx-cc13xx/rf-core/smartrf-settings.c#L252

arurke, 2017-07-05:
I believe you 😄