contiki-os/contiki #1662: Zolertia Z1 - Conflicting activation of phidget and battery sensors ?

seza443, 2016-05-13:
I build a small [game controller for the Z1](https://github.com/mokaddem/Z1_Mote_Controller).
I have a process reading values from a Joystick sensor. It works fine.

Then, I added a second process, reading the value of the battery sensor every 5 minutes. But it makes the Joystick stop working: the value does not update anymore!

I found a workaround: when I have to read the value of the battery, I deactivate the *phidget_sensor*, activate the *battery_sensor*, read the value and then deactivate the *battery_sensor* and reactivate the *phidget_sensor*.

But I would like to know **why** I can not have both sensors activated at the same time ?

Thanks

alignan, 2016-05-13:
Just check how the driver is done.
Basically each component (battery and phidget drivers) configure the ADC on its own.

seza443, 2016-05-15:
Hi @alignan 
Not really the answer I expected since I don't know what is the ADC :smile: and I have no desire to look at the driver :wink: 
Anyway, I'll close this issue and hope one day someone will answer to the [stackoverflow](http://stackoverflow.com/questions/37209097/contiki-os-on-zolertia-z1-conflicting-activation-of-phidget-and-battery-sensor) version of my question

Thanks

closed 2016-05-15 by seza443

alignan, 2016-05-15:
Well the ADC is the "analogue to digital converter", basically is the component that provides you the voltage signal levels of an analogue sensor, so then it can later be used to translate to a meaningful value.

What happens is the battery sensor driver and the phidget driver each when starts configures the ADC on its own, thus overwriting the ADC configuration.

The expected use of both of these components is actually how you are actually using: enable, measure, then disable.  This way you ensure at all times the ADC is configured the way your application expects.  If you want to have this done in a single operation then I'm afraid you will need to modify probably the phidget driver and include this.

I hope this is the answer you expected, as you are asking why does this happens.

seza443, 2016-05-15:
Awesome answer, exactly the one I was looking for! Many thanks :)
And I validated your answer on stackoverflow, thanks again.