contiki-os/contiki #1675: Is the length of packet attribute always less than 8 ?

tidyjiang8, 2016-05-24:
Hello all,
I'm reading the source code `core/net/rime/chameleon.c`.  The fuction `hdrsize` is used to calculate the total length of final header. It consider the sum of each attribute as header length, but the code is:
```
for(; a->type != PACKETBUF_ATTR_NONE; ++a) {
      ...
      len = a->len;
      if(len < 8) {    // 
        len = 8;
      }
    size += len;
}
```
It only deal with the length of less than 8? But the type of attribute is `typedef uint16_t packetbuf_attr_t;`, that is, the scope of `len` belongs to [0, 16]

I think it may be a bug, right ?

Thanks.
Tidy.

greg-king5, 2016-05-24:
It's the other way around; the `len` that's added to `size` is _never_ less than eight.

atiselsts, 2016-05-24:
Yes, I think this is a bug that in some marginal situations will cause incorrect behavior (with attribute lengths that are > 8 bits and not divisible by 8).

The correct way probably is to reuse the one-liner from `input` and `output` functions in the same file:

    len = (a->len & 0xf8) + ((a->len & 7) ? 8: 0);

This is quite a marginal bug in a marginal area of the OS, so I don't know if anyone cares.

tidyjiang8, 2016-05-25:
@greg-king5
Yes, I konw that all packet attributes' length isn't larger than 8, as follow:

|Primitive| Attribute| Bits |Scope|
| ------------- |:-------------:| -----:|-----:|
|ibc| Single-hop sender| Addr len| 1|
|uc| Single-hop receiver |Addr len |1|
|stuc| Retransmissions| 4 |1|
|ruc| Single-hop reliable |1| 0|
|ruc| Single-hop packet type |1| 1|
|ruc| Single-hop packet ID |2 |1|
|ruc| Max. rexmit |4| 0|
|rmh| End-to-end sender |Addr len| 2|
|rmh| End-to-end receiver| Addr len| 2|
|rmh| Time to live| 5| 2|

But in order to compatible with underlying protocal, it's necessary to deal with the length larger than 8,. I think that is also the reason why the `packetbuf_attr_t` is defined as uint16_t rather than uint8_t.

@atiselsts 
The expression `len = (a->len & 0xf8) + ((a->len & 7) ? 8: 0);` is also trouble me. It's a little bit tenebrous and unnecessary, because the scope of len is [0, 16]. `len = (a->len > 8) ? 16 : 8` is easier to understand.
Uh, I'm squeamish