contiki-os/contiki #1680: error message of rpl_tsch

sunnyzhuhai, 2016-05-25:
when i try to make the file inside the folder of "rpl_tsch", there are several error message.

sunnyzhuhai, 2016-05-26:
this is the error message:

../../../core/net/mac/tsch/tsch-slot-operation.c:104:2: error: #error "TSCH: RTIMER_SECOND < (32 * 1024)"
 #error "TSCH: RTIMER_SECOND < (32 * 1024)"
  ^
../../../core/net/mac/tsch/tsch-slot-operation.c: In function ‘tsch_tx_slot’:
../../../core/net/mac/tsch/tsch-slot-operation.c:472:95: error: ‘RADIO_DELAY_BEFORE_TX’ undeclared (first use in this function)
           TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_tx_offset] - RADIO_DELAY_BEFORE_TX, "TxBeforeTx");
                                                                                               ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro ‘TSCH_SCHEDULE_AND_YIELD’
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:472:95: note: each undeclared identifier is reported only once for each function it appears in
           TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_tx_offset] - RADIO_DELAY_BEFORE_TX, "TxBeforeTx");
                                                                                               ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro ‘TSCH_SCHEDULE_AND_YIELD’
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:479:11: warning: implicit declaration of function ‘US_TO_RTIMERTICKS’ [-Wimplicit-function-declaration]
           tx_duration = TSCH_PACKET_DURATION(packet_len);
           ^
../../../core/net/mac/tsch/tsch-slot-operation.c:501:102: error: ‘RADIO_DELAY_BEFORE_RX’ undeclared (first use in this function)
                   tsch_timing[tsch_ts_tx_offset] + tx_duration + tsch_timing[tsch_ts_rx_ack_delay] - RADIO_DELAY_BEFORE_RX, "TxBeforeAck");
                                                                                                      ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro ‘TSCH_SCHEDULE_AND_YIELD’
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:509:47: error: ‘RADIO_DELAY_BEFORE_DETECT’ undeclared (first use in this function)
               ack_start_time = RTIMER_NOW() - RADIO_DELAY_BEFORE_DETECT;
                                               ^
../../../core/net/mac/tsch/tsch-slot-operation.c: In function ‘tsch_rx_slot’:
../../../core/net/mac/tsch/tsch-slot-operation.c:666:89: error: ‘RADIO_DELAY_BEFORE_RX’ undeclared (first use in this function)
     TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_rx_offset] - RADIO_DELAY_BEFORE_RX, "RxBeforeListen");
                                                                                         ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro ‘TSCH_SCHEDULE_AND_YIELD’
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:680:38: error: ‘RADIO_DELAY_BEFORE_DETECT’ undeclared (first use in this function)
       rx_start_time = RTIMER_NOW() - RADIO_DELAY_BEFORE_DETECT;
                                      ^
../../../core/net/mac/tsch/tsch-slot-operation.c:766:19: warning: implicit declaration of function ‘RTIMERTICKS_TO_US’ [-Wimplicit-function-declaration]
                   &source_address, frame.seq, (int16_t)RTIMERTICKS_TO_US(estimated_drift), do_nack);
                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:780:73: error: ‘RADIO_DELAY_BEFORE_TX’ undeclared (first use in this function)
                   packet_duration + tsch_timing[tsch_ts_tx_ack_delay] - RADIO_DELAY_BEFORE_TX, "RxBeforeAck");

yatch, 2016-05-27:
I guess you hit `make` without any option under the directory, `rpl-tsch`. This causes the system to make binaries for the `native` platform, which does not support TSCH. 

You will find which platform works with TSCH in [README](https://github.com/contiki-os/contiki/blob/master/core/net/mac/tsch/README.md#features). For example, `z1` is one of such platforms. Try `make TARGET=z1` under the directory. The compilation will succeed. 

```C
user@instant-contiki:~/contiki/examples/ipv6/rpl-tsch$ make TARGET=z1
```

But, to compile the code is not your goal, is it? The best starting point for the TSCH implementation is to play `rpl-tsch-z1.csc` with Cooja.

alignan, 2016-06-01:
@sunnyzhuhai: does @yatch advice solved your problem?

saleemrazam, 2016-06-13:
Hello every one,

I have few quesions
1. Is it mandatory to have vmware to fun contiki and cooja or can it also be run directly on a ubuntu 14.04?

2. I have download this conitki-master folder form this website, how should i run it with vmware?

3. I Have tried  user@instant-contiki:~/contiki/examples/ipv6/rpl-tsch$ make TARGET=z1  on ubutnu 14.04 but i get the sam errors as @sunnyzhuhai has mantioned

PLease help

Thanks
Saleem Raza

yatch, 2016-06-15:
@saleemrazam No, it's not mandatory. But, using [Instant Contiki](http://www.contiki-os.org/download.html) with VMware is the easiest way to start with Contiki. All the requisite packages are installed into an image of Instant Contiki; it's ready to use. You might want to see [Get Started with Contiki](http://www.contiki-os.org/start.html).