contiki-os/contiki #1685: Unable to compile Cooja in Contiki-3.0

jimmytmathew, 2016-05-29:
BUILD FAILED
/home/user/Downloads/contiki-3.0/tools/cooja/build.xml:199: The following error occurred while executing this line:
/home/user/Downloads/contiki-3.0/tools/cooja/apps/mspsim/build.xml:29: -
----------------
Could not find the MSPSim build file. Did you run "git submodule update --init"?
----------------

jimmytmathew, 2016-05-29:
Got the solution. Check the link
https://sourceforge.net/p/contiki/mailman/message/34660437/

closed 2016-05-29 by jimmytmathew

adnanrashidpk, 2019-07-24:
The following worked for me:

Download the mspsim separately from this link https://github.com/contiki-os/mspsim (check your home/contiki/tools/mspsim and home/contiki-3.0/tools/mspsim directories, both are probably empty).
Remove the empty mspsim directory from both places
After unzip mspsim-master, paste the unzipped mspsim-master directory on both places
Rename the directory from mspsim-master to mspsim on both palces.
Go to terminal (user@instant-contiki:~/contiki-3.0/tools/cooja$) and run this command ant run
No need to use sudo
Enjoy!