contiki-os/contiki #1693: how to remove all routes from the network to a node ? Or how to discard a node completely

MalathiRK, 2016-06-01:
I need to know the API to remove a node from network. i.e, I need to completely remove all routes and thus discard the node.

Please help me, sir.

MalathiRK, 2016-06-01:
Sir,

Please help me in this problem.

I need to know the API to remove a node from network  completely.  I had used uip_ds6_nbr_rm() , but it does not work out. Is this API can remove the node which is directly connected to root? I need to remove the node irrespective of whether or not that node is directly related(child) related to DAG root. How can I do?

How can I get address of newly joined node.

Please help me ..

Thanks
Malathi

g-oikonomou, 2016-06-01:
https://github.com/contiki-os/contiki/wiki#Help

closed 2016-06-01 by g-oikonomou