contiki-os/contiki #1697: Cooja compilation errors due to "unmappable character for encoding ASCII"

viktpr, 2016-06-01:
Trying to compile Cooja results in a bunch of (92) errors of the following kind, all caused by files in tools/mspsim:

    [javac] /usr/home/viktor/src/bsc/practical/contiki/tools/mspsim/se/sics/mspsim/chip/I2CUnit.java:1: error: unmappable character for encoding ASCII
    [javac] /* Copyright (c) 2013, tado?? GmbH. Munich, Germany.
    [javac]    

I am a huge Java dummy so the only solution I came up with was removing the problematic characters, which is not a solution at all, esp. since mspsim is upstream code.
The other solution is much nicer but I believe someone who knows what they are doing can do it more elegantly:

     JAVA_TOOL_OPTIONS=-Dfile.encoding=UTF8 ant run

I don't know if it is relevant but I am using FreeBSD with:

    $ java -version
    openjdk version "1.8.0_77"
    OpenJDK Runtime Environment (build 1.8.0_77-b03)
    OpenJDK 64-Bit Server VM (build 25.77-b03, mixed mode)

viktpr, 2016-06-01:
Another solution I just found is adding the

   <compilerarg line="-encoding utf-8"/>

to the mspsim/build.xml "compile" target.

bernardobelchior, 2017-10-03:
I had the same problem and @viktuor's 2nd solution worked perfectly.