contiki-os/contiki #1700: mqtt with launchpad cc2650

saulofr, 2016-06-02:
@g-oikonomou I can connect the  launchpad with a MQTT broker by 6lbr IP64 config? I try change the ip broker using RFC 6052 prefix but it dont work.

xukai871105, 2016-06-02:
We have the same problem。When we use 6lbr and turn on the IP64 function, We find that UPD work well, But there is something wrong with TCP.Because of TCP error, We could not use MQTT client to connect a local borker or IBM Bluemix(IoT Waston Platform). We tried to fix the problem using CC2538dk and CC1310DK.

g-oikonomou, 2016-06-02:
I've used CC2650 + MQTT + NAT64 + Bluemix in the past, zero problems. The only difference I can see to what you are describing is that I was not using IP64 for NAT64, I was using wrapsix on a Beaglebone Black.

From what is being described here, this doesn't appear to be a problem with the LP or CC2650 specifically, most likely not a problem with MQTT either.

If UDP works but TCP does not, then we'll need more information in order to figure out why.

xukai871105, 2016-06-02:
We have tested CC1310DK + rpl_board_router + tunsilp6 + MQTT(or http-example) + NAT64(Jool).There is no problem in this situation. When we use 6lbr + slip-radio instead of rpl_board_router + tunslip6 a TCP error happens. CC1310dk can send TCP SYN at the right way but can not receive TCP SYN+ACK. HTTP server or MQTT broker tries to retransmit SYN+ACK but it has no effect. In this pcap file, CC1310dk IP address: 192.168.1.194.Http server IP address 192.168.1.104.
[image]
Please forgive my poor English.

g-oikonomou, 2016-06-02:
@laurentderu Could this be a problem in 6lbr, Laurent?

alignan, 2016-06-02:
Have you tried using the `::ffff` in `6lbr` instead of the `RFC6052`?

saulofr, 2016-06-02:
it is showing in the mapping. I'm using the mosca broquer , could this be?

[image]

saulofr, 2016-06-02:
@alignan, yes I try it, but not work, thanks

g-oikonomou, 2016-06-02:
Off-topic, but if the thing in the top-left is PuTTY, you should set "Implicit CR every LF" in this session's options (or was it "Implicit LF every CR"???)

saulofr, 2016-06-02:
CR in every LF, thanks!

closed 2016-06-26 by g-oikonomou