contiki-os/contiki #1707: Unsafe handling of strings in parse_url function of http-socket.c

vsaw, 2016-06-05:
As mentioned in https://github.com/contiki-os/contiki/pull/1696 the [`parse_url`](https://github.com/momote-io/contiki/blob/47df9c4e6ec6402a2be1bbf11c12ca0e63a0a9fc/core/net/http-socket/http-socket.c#L292) function is currently unsafe in it's handling of strings. It risks buffer overflows and might not set `\0` terminator of the strings it parses. Furthermore it does not return error codes to callers if the hostname of the url does not fit into the buffer.

The behaviour can be triggered by two things:

1. Buffer overflows will be triggered if the buffer of the `host` parameter is less than `MAX_HOSTLEN` long.
2. The `\0` terminator will not be set if the host name in the URL is longer than `MAX_HOSTLEN`

This makes a correct and save invocation of `parse_url` hard if the caller does not know these requirements. I therefore propose to change the following things about `parse_url`.

1. Explicitly give the buffer sizes as a parameter to `parse_url`
2. Always set the `\0` string terminator
3. Return with the error code `return 0;` if the host could not fit in the buffer.

casaroli, 2016-06-06:
IMO Number 3 is already fixed in #1696 since parse_url is already returning 0 if host could not be null terminated.

Maybe if we force the \0 termination in that case (and still return 0) can solve number 2.

would that be enough?

vsaw, 2016-06-06:
@casaroli: True, thanks for pointing this out. I like the idea to always terminate the host string 👍