contiki-os/contiki #1713: GPIO API missing

alignan, 2016-06-09:
More than an issue this is a call for discussion and comments.

CPU implementations define their own GPIO functions and calls, but I think we need to have an unified GPIO API, even if the current register-based macros are more efficient.  This will allow more portable implementations of sensors and actuators, as well as implement guidelines for people contributing their own platforms.

tonnenpinguin, 2016-06-09:
How about using the arduino GPIO API as a starting point?

alignan, 2016-06-09:
[rant] God no! [/rant]

Joking :smile:

bthebaudeau, 2016-06-09:
We would need that, not only for GPIO, but for all drivers. CMSIS aims at standardizing APIs for Cortex-M, but this could probably apply to other platforms too.

Regarding GPIOs, there are two parts:
 - GPIOs per se (i.e. sensing a pin or setting its level in output mode). The gpio API from U-Boot or Linux would be great here.
 - Pad control (i.e. muxing of alternate functions, open drains, keepers, pull-up/down resistors, slew rate, drive strength, etc.). This is very platform-specific. This is handled by pinctrl in U-Boot and Linux.

gebart, 2016-06-10:
We have been using the RIOT API for GPIOs [1] in our Mulle Contiki port [2] (as well as most other in-CPU peripherals) to reduce the maintenance burden and to make apps easily portable between the two systems. We have also modified the Makefiles to build the RIOT drivers in a separate archive so it is easily replaceable (to clearly show the intent to comply with the LGPL and its requirements on modifying and relinking).

[1]: http://riot-os.org/api/group__drivers__periph__gpio.html
[2]: https://github.com/eistec/contiki/blob/mulle-master/cpu/arm/k60/include/RIOT/periph/gpio.h

alignan, 2016-06-10:
Thanks @gebart! I have also reviewed the RIOT API while porting the `RE-Mote` and is OK.  The link to your Contiki port seems broken.

alignan, 2016-06-10:
@bthebaudeau I will check U-Boot also, I only used it briefly but having something close to Unix users seems a godd idea

greg-king5, 2016-06-12:
> Thanks, @gebart! I have also reviewed the RIOT API while porting the RE-Mote; and, it is OK.
> The link to your Contiki port seems broken.

Yes, he used an unusual style of writing URL links.  Then, he put a phrase in parentheses next to the second link.  Unfortunately, that made the phrase look like a standard Markdown link.  Here is his comment again, written in the standard Markdown style:

> We have been using the RIOT API for GPIOs [[1]](http://riot-os.org/api/group__drivers__periph__gpio.html) in our Mulle Contiki port [...]

gebart, 2016-06-13:
sorry about that, I have edited my comment above to fix the links