contiki-os/contiki #1717: Application Layer Protocols | Contiki OS v.3

kaleemullah360, 2016-06-10:
With reference to [PRISMTECH](http://www.prismtech.com/sites/default/files/documents/Messaging-Whitepaper-040615_1.pdf) white paper published in 2016 for iot messaging protocols.

I'm searching list of application layer protocols used in `Contiki OS`.
I easily find MAC layer protocols used in Contiki OS. i.e `Null MAC`, `Contiki MAC`, `X-MAC` and so on.
and found only three Application layer protocols.
A. [HTTP](https://en.wikipedia.org/wiki/Hypertext_Transfer_Protocol)
B. [MQTT](http://mqtt.org/)
C. [CoAP](http://tools.ietf.org/pdf/rfc7252.pdf)

so I was looking if Contiki OS ported other Application layer protocols.

1. [AMQP](http://www.amqp.org/)
2. [XMPP](http://xmpp.org/)
3. [DDS](http://portals.omg.org/dds/)
4. [STOMP](https://stomp.github.io/stomp-specification-1.2.html)
5. [JMS](https://en.wikipedia.org/wiki/Java_Message_Service)
6. [SMQ](https://realtimelogic.com/ba/doc/?url=SMQ.html)
7. [AllJoyn](https://allseenalliance.org/developers/learn)
8. [HTTP/2](https://http2.github.io/)

please guide me about these protocols. and which of these are being used in Contiki OS. how can I port them ?
any help/guide resources ?

PY1CX, 2016-06-12:
Hi kaleemullah360, did you saw this paper from Dunkels? 
http://dunkels.com/adam/kovatsch11low-power.pdf CoAP + Contiki

kaleemullah360, 2016-06-12:
yes I already viewed the paper by  Dunkels. they published it when they ported CoAP protocol in Contiki OS. now looking for rest of 8 listed protocols.

closed 2016-07-22 by alignan