contiki-os/contiki #1722: CC2650 Sensortag Web Demo continuously rebooting

Rasfumi, 2016-06-12:
After pulling in the latest changes from contiki master branch the web-demo example is continuously rebooting. After disabling the watchdog i found out that after about 20 seconds the CPU gets stuck in FaultISR(). It seems it has something to do with the Pull request #1684. As this happens when trying to access a power domain that is disabled. In this case the AUX PD is off as i can see from Registers view.

In my own application that worked before the CPU gets stuck in the while loop of:
AuxAdiDdiSafeRead(uint32_t nAddr, uint32_t nSize)
{
    uint32_t nRet;
    // Disable interrupts and remember whether to re-enable
    bool bIrqEnabled = !CPUcpsid();
    // Acquire semaphore for accessing ADI/DDI in AUX, perform access, release semaphore
    while (!HWREG(AUX_SMPH_BASE + AUX_SMPH_O_SMPH0));

After calling "oscillators_switch_to_hf_xosc()" from ieee-mode.c on();
In this case the HW-Semaphore seems to be locked by another resource.

g-oikonomou, 2016-06-12:
Hi @Rasfumi 

I see you have deleted some stuff from the OP compared to your original post, something about your own application that used to work. As such, I cannot immediately understand whether you are using the example as-is, or whether you have local modifications. Have you made any changes whatsoever to the example?

I am now trying with SmartRF + CC2650EM and I cannot see anything strange.

g-oikonomou, 2016-06-12:
I'm not seeing any problems with the sensortag either. I'll let it run for a while though.

Rasfumi, 2016-06-13:
Hi @g-oikonomou 
thank you for your fast reply. In my own Application I found out it was my mistake that led to that infinite loop waiting for the HW Sempaphore.  I am using the Sensorcontroller and registered aux_ctrl consumer too late. For comparing to a working program i downloaded a fresh copy of contiki and tried the web-demo example which led to the result described. I only commented out the watchdog_start() in contiki-main.c to see where it hangs.
I just tried it again with the same result. Build command: make clean TARGET=srf06-cc26xx BOARD=sensortag/cc2650 cc26xx-web-demo.bin
It starts blinking fast, then connects to my 6lbr, blinks slower then stops blinking for a few seconds before rebooting.  (watchdog enabled again)
I am using TI CodeComposer  and compiler "gcc-arm-none-eabi-4_9-2015q3" with the DebugDevPack. A setup which used to work before.

g-oikonomou, 2016-06-13:
Please, can you try with 5.2.1 20151202 (the one used by travis)

https://github.com/contiki-os/contiki/tree/master/platform/srf06-cc26xx#requirements

Rasfumi, 2016-06-13:
I tried using the specified compier (5.2.1 20151202) and it really makes a difference, i even cross checked it. No more rebooting now. BUT with the newer compiler i cant access the sensortag through COAP or Web even it shows up in 6lbr. Other motes with older versions of contiki can be accessed though. Very strange...

g-oikonomou, 2016-06-13:
Smells like this might be unrelated. Are you using latest 6lbr too? I suspect this could have to do with 6lo address compression contexts.

Rasfumi, 2016-06-13:
i am using 6lbr 1.4.0 on raspberry PI connected to a CC2538-CC2592 Board with a slightly modified (RF-Amp Control) contiki version (latest commit 2016-04-07).

Rasfumi, 2016-06-14:
I agree that the connection problem has nothing to do with #1684. Anyway its a strange behavior and i still have no idea what causes it. I have updated my slip-radio with the latest contiki, but as expected this didnt help. Lets see if other users have the same problem.
Thanks alot for your help and the work you are doing for the community!

g-oikonomou, 2016-06-14:
See also #1573 and #1593, these could help.

closed 2016-06-14 by g-oikonomou