contiki-os/contiki #1733: Disabling DIO suppression

vamsi-chikati, 2016-06-21:
In the file rpl-timers.c at line 175

Shouldn't the condition to check DIO suppression be like this 

if((instance->dio_redundancy != 0 && instance->dio_counter < instance->dio_redundancy) || (instance->dio_redundancy == 0)) {
#if RPL_CONF_STATS
      instance->dio_totsend++;
#endif /* RPL_CONF_STATS */
      dio_output(instance, NULL);
    } else {
      PRINTF("RPL: Suppressing DIO transmission (%d >= %d)\n",
             instance->dio_counter, instance->dio_redundancy);
    }

Instead of 

if(instance->dio_redundancy != 0 && instance->dio_counter < instance->dio_redundancy) {
#if RPL_CONF_STATS
      instance->dio_totsend++;
#endif /* RPL_CONF_STATS */
      dio_output(instance, NULL);
    } else {
      PRINTF("RPL: Suppressing DIO transmission (%d >= %d)\n",
             instance->dio_counter, instance->dio_redundancy);
    }

In the present code, if REDUNDANCY is set to 0, DIO is always supressed!!!

Or am I missing something?

laurentderu, 2016-06-21:
The magic value 0 means the DIO suppression mechanism is disabled, this is required by RPL RFC and Trickle RFC :

<pre>
   k:    learned from the DIO message as DIORedundancyConstant.  The
         default value of DIORedundancyConstant is
         DEFAULT_DIO_REDUNDANCY_CONSTANT.  In RPL, when k has the value
         of 0x00, this is to be treated as a redundancy constant of
         infinity in RPL, i.e., Trickle never suppresses messages.
</pre>

greg-king5, 2016-06-21:
Two tests of `dio_redundancy` are not needed.  The line should be:
```c
if(instance->dio_redundancy == 0 || instance->dio_counter < instance->dio_redundancy) {
```

vamsi-chikati, 2016-09-30:
Duplicate of #1879

closed 2016-09-30 by vamsi-chikati