contiki-os/contiki #1735: Coffee: Shrink header with !COFEE_MICRO_LOGS

arthurfabre, 2016-06-22:
Micro logs can be disabled by setting `COFFEE_MICRO_LOGS` to 0. This removes most but not all the micro log related code (bits and pieces in `merge_log() ` are still around), but doesn't remove the log related parameters from the file headers (6 bytes).

Would there be interest in a pull request(s) that:

1. Remove all the micro log related code when `!COFFEE_MICRO_LOGS`
2. Removes the log parameters from the header (perhaps with another #define, eg `COFFEE_SMALL_HEADER`)

nvt, 2016-06-27:
Nice to see some suggestions to improve Coffee's configurability. In these two cases, however, it does not seem that the space savings are enough to motivate the added conditionals, which will make the source more difficult to read. In the second case, it will be more difficult to use the same storage (or image of the storage) with different Coffee configurations unless other header modifications are made.

Have you measured how much code space you save doing the first modification?

arthurfabre, 2016-06-30:
I haven't checked, but I suspect that very little code space is saved in the first case - this was mostly done to be able to implement the second case.

The second modification reduces Coffee file headers by 6 bytes (out of 10 bytes + `COFFEE_NAME_LENGTH`.