contiki-os/contiki #1744: Example cc26xx-demo Continually Reboots with Latest Contiki

davidjnix, 2016-06-24:
The cc26xx-demo on the SmartRF06 Eval Board reboots every 12 seconds or so. I am using the latest pull (f0b19a1). There was a similar report (#1465) from January 2016 that was resolved at that time. I verified that the fix for #1465 is in my repo. The serial output below demonstrates the reboot.

Starting Contiki-3.x-2647-gf0b19a1
With DriverLib v0.44336
TI SmartRF06EB + CC26xx EM
 Net: sicslowpan
 MAC: CSMA
 RDC: ContikiMAC, Channel Check Interval: 16 ticks
 RF: Channel 25
 Node ID: 34057
CC26XX demo
-----------------------------------------
Bat: Temp=25 C
Bat: Volt=3292 mV
ALS: 687 raw
Starting Contiki-3.x-2647-gf0b19a1
With DriverLib v0.44336
TI SmartRF06EB + CC26xx EM
 Net: sicslowpan
 MAC: CSMA
 RDC: ContikiMAC, Channel Check Interval: 16 ticks
 RF: Channel 25
 Node ID: 34057
CC26XX demo
-----------------------------------------
Bat: Temp=25 C
Bat: Volt=3292 mV
ALS: 685 raw

g-oikonomou, 2016-06-26:
What toolchain version are you using?

g-oikonomou, 2016-06-26:
I can confirm this. I suspect it's something to do with #1709

g-oikonomou, 2016-06-26:
Can you try with `#define CC2650_FAST_RADIO_STARTUP 0`?

You can change the respective line in `contiki-conf.h` or set it in `project-conf.h`

davidjnix, 2016-06-27:
George,

Will do later today. Thanks for the quick response.

David

davidjnix, 2016-06-27:
That fixed the issue. Verified using (f0b19a1) with SmartRF06 Eval Board and toolchain gcc-arm-none-eabi-5_2-2015q4.

PR #1746 (2016-06-28): Disable CC2650_FAST_RADIO_STARTUP by default
`CC2650_FAST_RADIO_STARTUP` is required for TSCH, but it currently breaks ContikiMAC. We set the default value to the one that will work with the default choice of RDC.

I'm also moving the define to a place in `contiki-conf.h` closer to where the user will look in order to configure the RDC driver.

Fixes #1744
commit 3a105f21 (2016-06-28): Disable CC2650_FAST_RADIO_STARTUP by default

`CC2650_FAST_RADIO_STARTUP` is required for TSCH, but it currently breaks ContikiMAC. We set the default value to the one that will work with the default choice of RDC.

Fixes #1744
closed 2016-06-28 by g-oikonomou