contiki-os/contiki #1758: CC2538 LPM breaks UART interrupts

arthurfabre, 2016-06-30:
With `LPM_CONF_ENABLE` set to `1`, UART interrupts are not reliably serviced by the ISR.

This can be observed using the following test case:

```
#include "contiki.h"
#include "sys/etimer.h"
#include "dev/uart.h"
#include <stdio.h>

volatile int count = 0;

static int uart_handler(unsigned char b) {
    count++;
    return 0;
}

PROCESS(uart_test, "UART Test");
AUTOSTART_PROCESSES(&uart_test);

PROCESS_THREAD(uart_test, ev, data) {
    static struct etimer et;
    PROCESS_BEGIN();

    // Disable the FIFO buffer
    REG(UART_0_BASE + UART_LCRH) &= ~UART_LCRH_FEN;

    uart_set_input(0, uart_handler);

    etimer_set(&et, CLOCK_SECOND * 10);

    while (1) {
        PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));

        printf("Intrs: %d\n", count);

        count = 0;

        etimer_reset(&et);
    }
  
    PROCESS_END();
}
```

With LPM enabled, no matter how many characters are sent to UART0, `count` remains 0.
With LPM disabled, `count` is consistent with the number of characters sent across.

bthebaudeau, 2016-06-30:
This is the expected behavior. From PM1, the CC2538 may only wake up from:
 - pin interrupts,
 - Sleep Mode timer,
 - USB resume.

With LPM enabled, the MCU can either be active or sleeping at various depths. If the UART interrupt occurs during active mode, sleep mode or PM0, it works, but it is missed if it occurs during PM1 or PM2. It can not even be delayed because there is no UART clock active to detect any incoming data and trigger an interrupt flag.

So if you need the UART interrupt, you either have to set `LPM_CONF_ENABLE` to `1`, or to set `LPM_CONF_MAX_PM` to `0`, which would at least allow sleep mode and PM0.

We could also introduce an API in the UART driver to disable PM1 during some time period during which the interrupt is needed, rather than having to disable PM1+ at compile time. But actually, you can already do that by calling `lpm_set_max_pm(0)`, then `lpm_set_max_pm(LPM_CONF_MAX_PM)` when you no longer need the UART interrupt.

Can you confirm that this fixes your issue?

arthurfabre, 2016-07-01:
Thanks for the detailed explanation - I can confirm that calling `lpm_set_max_pm(LPM_PM0)` and `lpm_set_max_pm(LPM_CONF_MAX_PM)` to temporarily disable PM1 when we expect UART interrupts works.

closed 2016-07-01 by arthurfabre