contiki-os/contiki #1766: TSCH: TSCH queue not fully flushed after disassociation

thomas-ha, 2016-07-05:
Hello,

I tested TSCH over a period of days and weeks with about 20 nodes and in conditions where the RF signal strength can vary pretty much. So sometimes a lot of packets get lost and nodes will disassociate. The nodes are sending packets to the border router constantly about every 20 seconds.
I have seen that it happens that at some point in time nodes don't re-associate because the TSCH queue got full and they don't send packets anymore:

`TSCH-queue:! add packet failed: 0 #0x20003bdc 2 #0x0 #0x0
TSCH:! can't send packet !tsch_queue_add_packet`

Otherwise the node is working correctly.

In my tests I found out that this happens when the node dis- and re-associates a lot of times. At each dis-association the tsch queue is flushed and reset. But at neary each dis-association the size of "packet_memb" in tsch-queue.c is reduced by 1. I printed out

`memb_numfree(&packet_memb)`

to see this. So after a couple of dis-associations I can see this in the TSCH output:

TSCH: {asn-0.157b link-1-29-18-1 ch-20} ! leaving the network, last sync 4704
TSCH-queue:! flushing packet
tsch_queue_free_packet: PACKET_MEMB NUMFREE: 1 
tsch_queue_reset: PACKET_MEMB NUMFREE: 1 
tsch_queue_reset: PACKET_MEMB NUMFREE: 1 
TSCH-queue:! flushing packet
tsch_queue_free_packet: PACKET_MEMB NUMFREE: 2 
tsch_queue_reset: PACKET_MEMB NUMFREE: 2 
TSCH: update time source: 1 -> 0
TSCH-schedule: remove_link 0 2 1 0 255
TSCH-schedule: remove_link 1 2 1 1 255
TSCH:! not associated, drop outgoing packet

In the beginning the size was 8:

tsch_queue_init: PACKET_MEMB NUMFREE: 8 

I couldn't find out why this happens yet.

simonduq, 2016-07-05:
Thanks for the detailed report! Will look into this at some point.

Conrad2210, 2017-02-09:
Is there any update on this issue? I have a similar problem and was wondering if a solution is available?

yatch, 2017-02-14:
@thomas-ha PR/https://github.com/contiki-os/contiki/pull/2046 could resolve the issue if its root cause is the same as discussed in PR/https://github.com/contiki-os/contiki/pull/2108.