contiki-os/contiki #1774: Run sniffer on CC1310 LaunchPad

chenek, 2016-07-14:
I try to build and run sniffer on CC1310 on LanuchPad but it seems not work. I also tried it with my CC1310DK and it doesn't work too. Does anyone can run CC1310 sniffer successfully with latest source code?

g-oikonomou, 2016-07-16:
@chenek Yes, it is possible to run CC1310 LP sniffer. In fact, any board/chip that uses the PROP mode driver can be used in this fashion.

You basically build the standard sniffer example under `examples/cc2538-common/sniffer` and you apply the changes below:

```
#define PROP_MODE_CONF_SNIFFER             1
#define CC26XX_UART_CONF_BAUD_RATE    460800
```

You can change the respective lines in the platform's `contiki-conf.h` or you can add them to the example's `project-conf.h`. Changing the UART speed is not strictly required, but it's recommended. Don't forget to run sensniff with the correct `-b` argument (`-b 460800`) in our case.

I'll document this in the platform's readme.

chenek, 2016-07-17:
Just verify this and find it works now. Thanks for your help.

closed 2016-07-17 by chenek

Arepas, 2016-11-03:
Hi, I'm newbie here I'm trying to do the same and use the CC1310 LP as sniffer. Where is supposed to be the code "#define PROP_MODE_CONF_SNIFFER 1"?

thanks you in advance :)

chenek, 2016-11-04:
There are some changes to build sniffer in Contiki. You can try to refer to https://github.com/contiki-os/contiki/tree/master/examples/sensniff