contiki-os/contiki #1788: CFS write may returns invalid length

mehulhirpara, 2016-07-18:
As per CFS write API signature, it returns "int" value which can be either negative value or number of bytes  written to the file system. API except input length in "unsigned int", which means 4GB of data. Now, if user calls API with value which is more INT_MAX, API may fail as it can accommodate only 2GB of data length.

CCIF int cfs_write(int fd, const void *buf, unsigned int len);

gebart, 2016-07-18:
This is also an issue in the specification of POSIX read/write, which CFS mimics.

http://pubs.opengroup.org/onlinepubs/009695399/functions/write.html
>If the value of nbyte is greater than {SSIZE_MAX}, the result is implementation-defined.

mehulhirpara, 2016-07-18:
Got it. I actually referred GNU standards @ https://www.gnu.org/software/libc/manual/html_node/Block-Input_002fOutput.html and got confused. Here definition of write is as follow:

size_t fwrite (const void *data, size_t size, size_t count, FILE *stream)

greg-king5, 2016-07-19:
That is `fwrite()`, a high-level function.  But, `write()` is a low-level function.  Look at https://www.gnu.org/software/libc/manual/html_node/I_002fO-Primitives.html

mehulhirpara, 2016-07-20:
Agree. write() is a system call and CFS has mimicked it to high level function cfs_write() and therefore it carries same signature.

Thanks

closed 2016-07-20 by mehulhirpara