contiki-os/contiki #1791: CC2650 LauchPad or CC2650STK+CC-DEVPACK-DEBUGGER cannot work as Contiki Sniffer

chenek, 2016-07-19:
I tried to make CC2650 LauchPad or CC2650STK+CC-DEVPACK-DEBUGGER work as Contiki Sniffer but cannot make it. I build sniffer under \contiki\examples\cc2538-common\sniffer by doing "make TARGET=srf06-cc26xx BOARD=launchpad/cc2650 sniffer.bin" for CC2650 LaunchPad and "make TARGET=srf06-cc26xx BOARD=sensortag/cc2650 sniffer.bin" for CC2650STK+CC-DEVPACK-DEBUGGER. After the sniffer.bin is built and download to my CC2650LP or CC2650STK, it doesn't capture anything. I test the same code base with CC2538DK and it works fine. Does anyone see similar issue?

g-oikonomou, 2016-07-19:
Hi @chenek I've used STKs as sniffers in the past.

I'll assume you are using sensniff. Make sure to run it with the correct `-b` argument: Default is 115200, but I strongly suggest setting the UART baud rate to 460800 and running sensniff with `-b 460800`. You can run it with with `-D INFO`, in which case it will print out more info. If you start seeing lines starting with "Peripheral: " followed by junk chars, you are most likely in the wrong baud rate.

For CC2538DK, the sniffer example defaults to 460800 UART baud rate on the peripheral side.

Lastly, make sure your sniffer is tuned in the correct channel.

chenek, 2016-07-19:
No, I don't see any junk chars when I use CC2650LP or CC2650STK. However, I would see chars output on UART when I use CC2538EM immediately. Can you test sniffer with your CC2650LP or CC2650STK using latest Contiki code?

g-oikonomou, 2016-07-20:
ugh (headbang), I just clicked. The IEEE mode RF driver does not support sniff mode (and this was a conscious decision). I was using prop-mode as sniffers in the past...

chenek, 2016-07-20:
Does this mean that we cannot use CC2650 as sniffer?

g-oikonomou, 2016-07-20:
Funny how you opened this thread just as I was about to push this...: https://github.com/g-oikonomou/contiki/tree/contrib/sensniff/examples/sensniff

So why don't you try this one out (tested with CC2650LP), with latest sensniff, and let me know what you think? :)

chenek, 2016-07-20:
I would like to test this but how can I git pull this commit?

g-oikonomou, 2016-07-20:
you can git remote add my repo and git fetch it, and then check out the corresponding branch.

chenek, 2016-07-20:
Sorry, I am not familiar with get remote add and git fetch. I have git Contiki to my PC and can you show me how to do it?

g-oikonomou, 2016-07-20:
```
git remote add <give-me-some-friendly-name> https://github.com/g-oikonomou/contiki
git fetch <give-me-some-friendly-name>
git checkout contrib/sensniff
```

Your local copy should now be showing you my branch.

chenek, 2016-07-20:
Thanks a lot. CC2650 LaunchPad works now but I still cannot make my CC2650STK work. Any idea?

g-oikonomou, 2016-07-20:
I cannot immediately think of anything, the code is cpu-specific (rather than board-specific), so it should have "just worked". Make sure your tag is printing things to start with. If you are using a debugger devpack, make sure the corresponding define is set to 1 in contiki-conf.h.

chenek, 2016-07-20:
It works now. I made the UART TX/RX pins wrong. Thanks for the help!

closed 2016-07-20 by chenek