contiki-os/contiki #1808: Persistent Travis failure

g-oikonomou, 2016-08-02:
Started appearing after merging #1493 (the test for the PR did pass though).

This is persistent: ROM overflow for the Z1 target for a couple of tests.

simonduq, 2016-08-03:
Seems to be a linker bug with msp430x: compilation on z1 fails with "2 bytes ROM overflow" now and then. Sometimes, even making any string *longer* by a few bytes solves it.
The fix is probably to do theses tests on another platform than z1, let me try this out.

simonduq, 2016-08-03:
So, I don't have a fix. Tried with exp5438 which is more roomy, but the problem also shows up. It seems the conditions to make it happen include:
* timerb defined
* far.rom used
* then depending on the "rom" size sometimes "data" overlaps by "vector" by 2 bytes, sometimes everything is fine.

Looks like an alignement error. Looked at the linker scripts but couldn't find the root cause.

What I suggest is (1) change the TSCH compile tests to a non msp430x platform and (2) disable the TSCH Cooja tests as long as the problem is around

simonduq, 2016-08-03:
Maybe once https://github.com/contiki-os/contiki/pull/1661 is merged we can try TSCH on Cooja motes instead :)

simonduq, 2016-08-03:
See https://github.com/contiki-os/contiki/pull/1809

closed 2016-08-03 by g-oikonomou