contiki-os/contiki #1817: Huge clock drift on CC2538-based platforms

kkrentz, 2016-08-10:
I am having trouble with the rtimer module on OpenMotes: I encountered a huge clock drift of around 0.5ms per s. Re-Motes seem to suffer from this problem, as well. It would be interesting for me if someone confirmed this finding. It would also be helpful if someone had suggestions on fixing this issue.

I have put together a conscious example here:
https://github.com/kkrentz/contiki/tree/drift
You can run it as follows:
```
cd examples/secmac/tests/drift
make drift.upload PORT=/dev/ttyUSB0 && grabserial -d "/dev/ttyUSB0" -b 115200 -t
```
and calculate the clock drift using this R-script:
```R
real_diff <- 253.152868 - 1.026931
mote_diff <- (8309108 - 51572)/32768
drift <- (real_diff - mote_diff)/real_diff
print(drift)
```

peretuset, 2016-08-10:
Do you have LPM2 enabled to save energy on the CC2538? This can add up at least 136 microseconds unless it is compensated by the code (i.e. wake ahead of time), which I am unsure. 
Please try to disable LPMx completely and try again.

kkrentz, 2016-08-10:
Indeed, the LPM module influences the clock drift. With `#define LPM_CONF_ENABLE 0`, I'd just obtained a drift of 3ms/s. Interestingly, with `#define LPM_CONF_MAX_PM 0` and `#define LPM_CONF_ENABLE 1`, I'd just obtained a clock drift of 0.25ms/s.

bthebaudeau, 2016-08-10:
LPM had an influence in the past, but it should no longer be the case. Which version of Contiki are you using? The rtimer now runs on the Sleep Mode Timer, which is only stopped in PM3, but not down to PM2.

With the master Contiki, the issue could be one of these:
 - the board has a 32,000-Hz crystal instead of a 32,768-Hz one,
 - the crystal capacitors have wrong values,
 - the board is running on the internal RC oscillator instead of on a 32,768-Hz crystal, either because of the software configuration (see `SYS_CTRL_CONF_OSC32K_USE_XTAL`), or because there is no such crystal on the board, in which case the accuracy is terrible, besides a systematic clock drift, which is not compensated by the software because it's a bit pointless to try improving the accuracy of an RC oscillator (but that could be done if this is really needed).

There have also been some recent cross-platform changes in the rtimer infrastructure that may be involved.

Also note that only a single rtimer may be used in the system.

joakimeriksson, 2016-08-11:
LPM will influence the clock drive quite significantly. Then only time the CC2538 syncs up the clock speed of the used clock is when it leaves LPM and wakes up from sleep. If you use this clock and do not sleep - it is not good for anything that requires a accurate timer. On platforms without the 32 768 crystal the clock system will likely need to be reconfigured when not sleeping at all. We did that at with some CC2538 based devices at Yanzi and got reasonable accuracy.

kkrentz, 2016-08-11:
Thanks for your quick responses!
@bthebaudeau You gave me the right hint. The crystal oscillator of OpenMotes is not used by default. Adding
```c
#undef SYS_CTRL_CONF_OSC32K_USE_XTAL
#define SYS_CTRL_CONF_OSC32K_USE_XTAL 1
```
reduced my clock drift to about 50ppm.
@peretuset Does the crystal oscillator run at 32,000Hz or 32,768Hz?

peretuset, 2016-08-11:
The crystal on the OpenMote is 32.768 kHz with a 30 ppm drift rate. I would say that your results are quite what they should be now.

closed 2016-08-11 by kkrentz