contiki-os/contiki #1829: question: cc26xx-web-demo-make error- new external sensor to the cc2650 sensortag

shosant, 2016-08-19:
Hi, 
I am extending the application cc26xx-web-demo to add an external sensor.
while compiling, I get the following error for the driver file of the external sensor.
Could somebody give some guidance:
co2-t6713-sensor.c is the driver file for the external sensor(i2c). 
I have followed the opt-3001-sensor.c to develop.

_**In file included from ../../../platform/srf06-cc26xx/sensortag/co2-t6713-sensor.c:40:0:
../../../core/./lib/sensors.h:47:14: error: expected declaration or statement at end of input
 const struct sensors_sensor name = { type, value, configure, status }
              ^
../../../platform/srf06-cc26xx/sensortag/co2-t6713-sensor.c:400:1: note: in expansion of macro 'SENSORS_SENSOR'
 SENSORS_SENSOR(co2_t6713_sensor, "CO2T6713", value, configure, status);
 ^
../../../platform/srf06-cc26xx/sensortag/co2-t6713-sensor.c:400:16: warning: unused variable 'co2_t6713_sensor' [-Wunused-variable]
 SENSORS_SENSOR(co2_t6713_sensor, "CO2T6713", value, configure, status);
                ^
../../../core/./lib/sensors.h:47:29: note: in definition of macro 'SENSORS_SENSOR'
 const struct sensors_sensor name = { type, value, configure, status }
                             ^
../../../core/./lib/sensors.h:47:14: warning: control reaches end of non-void function [-Wreturn-type]
 const struct sensors_sensor name = { type, value, configure, status }
              ^
../../../platform/srf06-cc26xx/sensortag/co2-t6713-sensor.c:400:1: note: in expansion of macro 'SENSORS_SENSOR'
 SENSORS_SENSOR(co2_t6713_sensor, "CO2T6713", value, configure, status);
 ^**_

Another question:
Can I use the bool_i2c_read( ) function from the platform/srf06-cc26xx/sensortag/board-i2c.c, to read my sensor?(in a similar way as used in the platform/srf06-cc26xx/sensortag/hdc-1000-sensor.c.

**_../../../platform/srf06-cc26xx/sensortag/co2-t6713-sensor.c:132:9: warning: implicit declaration of function 'bool_i2c_read' [-Wimplicit-function-declaration]
         success = bool_i2c_read((uint8_t *)&status_readbuf[4],4);_**

Thank you in advance.

greg-king5, 2016-08-19:
You have a typo in the function above that `SENSORS_SENSOR()` line.  It confused the compiler.  That's why it said, "control reaches end of non-void function".  (The other messages are just consequences of that "control ..." error.)

---
> Another question:
>  Can I use the `bool_i2c_read()` function from the "platform/srf06-cc26xx/sensortag/board-i2c.c", to read my sensor (in a similar way as used in the [...]

Those functions return a `bool` value; but, their name prefix is `board_i2c_`, not `bool_i2c_`.

shosant, 2016-08-22:
thanks a lot for pointing out those silly mistakes.
Have fixed them.

I get the **error: invalid storage class for function** , when I compile if the functions are declared as static  (in the driver file of the sensor which I created for my sensor). If it's not static I am able to compile the code.
Why is that so?
Can I use the it without static?

shosant, 2016-08-24:
Does building the code in a different version can cause problems?
mine is:
 _**arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 224288]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE**_.

My sensortag, keeps rebooting whenever I try connect to the webpage of the sensortag address.
http://[fd00::212:4b00:7a8:d606]/.. and I get page loading error.
If I don't try to access the page, the sensortag is stable (Led blinks slowly, I am able to see the sensortag in the 6lbr page,coap is also loaded.) But once I click the "Web" link of the sensoetag it reboots.
Also, I get a warning in the 6lbr error log, Web buffer overflow and eventually SLIP device times out.
Should I increase any buf size in my code, or the problem could be in driver file?

Any tips on troubleshooting the issue will be helpful, as I'm new to development.
Thank you.

g-oikonomou, 2016-08-27:
Does the sensortag also reboot with the version of the code on master without your changes?

shosant, 2016-09-11:
Thank you for the reply. Rebooting was because my code was pointing to contrib/sensiff. Once changed to master it was fine.

I have a issue with the datatype.
The sensortag reads the data from the co2 sensor. The values will be 
0x04
0x02
data_msb
data_lsb.
I'm declaring the type uint8_t.
When I try to print the msb or lsb value they are wrong . I get some random value printed such as 113326 (d) or in hex 1BAA0 etc. 
But it should be a 8 bit value.
The first 2 values are correct. When I print them I get 04 or 02 and combined 16bit value is 0402.
Could there be any problem with my type casting? My module for reading the data:

static bool
read_data(uint16_t *raw_co2gasppm, uint16_t *raw_co2gasppm_msb, uint8_t *raw_co2gasppm_lsb )
{
      bool success;
      /* uint8_t co2gasppm_databuf[4];
      uint8_t co2gasppm_requestbuf[5] = {0x04,0x13,0x8B,0x00,0x01}; */

      select_on_bus();

      success = board_i2c_write((uint8_t *)&co2gasppm_requestbuf,6);

      if(success)
      {  
        timer_set(&t, CLOCK_SECOND/10); /* timer for 50 ms */
        if(timer_expired(&t))
        {
        success = board_i2c_read((uint8_t *)&co2gasppm_databuf, 5); /*board_i2c_read((uint8_t *)&co2gasppm_databuf,4);*/
        }
        timer_reset(&t);
     }else
     {
         return CC26XX_SENSOR_READING_ERROR;
     }
     if(success)
     {
          uint8_t a = co2gasppm_databuf[0];
          uint8_t b = co2gasppm_databuf[1];
          uint8_t c = co2gasppm_databuf[2];
          uint8_t d = co2gasppm_databuf[3];
          if(a == 0x04 && b == 0x02) /* sanity check */
          {
               *raw_co2gasppm = ((uint16_t) c << 8) | ((uint8_t)d);
              *raw_co2gasppm_msb = c ;
              *raw_co2gasppm_lsb = d ;
          }else
          {
              return CC26XX_SENSOR_READING_ERROR;
          }
      deselect_on_bus();
     }else
     {
         return CC26XX_SENSOR_READING_ERROR;
     }
     return success;
}
static int
value(int type)
{
  bool success;
  int rv; 
  
  uint8_t converted_val;
  uint8_t rawdata_co2gasppm_msb;
  uint8_t rawdata_co2gasppm_lsb;
  uint16_t rawdata_co2gasppm;
   
  
   success = read_data(&rawdata_co2gasppm, &rawdata_co2gasppm_msb, &rawdata_co2gasppm_lsb);
    
   if(success) 
  { 
    converted_val = rawdata_co2gasppm_lsb ;

    PRINTF("GASPPM: %04X            g=%04X (ppm)\n",rawdata_co2gasppm,
         converted_val );

   rv = (int) converted_val ;
    
  } else
  {
return CC26XX_SENSOR_READING_ERROR;
  }

  return rv;
}

shosant, 2016-09-11:
Where can I  check the printf statements executed?

closed 2017-01-15 by g-oikonomou