contiki-os/contiki #1831: Issue while flashing code to cc1310F64

lovelesh, 2016-08-21:
Hi All,

I am trying to program cc1310F64 (64 KB Flash version). I am getting a peculiar error with the motes.

For **CC1310F128** in Contiki:
Bootloader configuration **CCFG.BL_CONFIG** is located at **0x0001FFD8** (Obtained through cc2538-bsl)

For **CC1310F64** in Contiki:
 Bootloader configuration **CCFG.BL_CONFIG** is located at **0x0000FFD8** (Obtained through cc2538-bsl)

Because of this issue I am not able to program the 64KB Versions.

lovelesh, 2016-08-22:
Contiki code requires 131KB of FLASH size, which is not possible for CC1310F64.

lovelesh, 2016-08-27:
Is there a mechanism to support multiple flash versions of the same chip?
For Example, if a chip supports 256KB and 512KB flash memories, then can it be supported under contiki?

g-oikonomou, 2016-08-27:
It won't be too difficult to add support for the CC1310 32 and 64KB flash variants. I am not aware of 256 and 512 versions.

lovelesh, 2016-08-27:
@g-oikonomou 

>  if a chip supports 256KB and 512KB flash memories
I meant chip like CC2538 or STM based MCU having higher FLASH

g-oikonomou, 2016-08-27:
I thought we were discussing CC13xx here :)

lovelesh, 2016-08-27:
I wanted to discuss generic mechanism for all platforms. but cc13xx can be a good starting point.

wubai, 2016-10-13:
@g-oikonomou 
So, how to add support for the CC1310 32 or 64KB flash variants?  Could you give me any advices?

lovelesh, 2016-10-13:
@wubai 
If you see the working (128KB version) and non-working (64KB) in the fist message. The only difference you will find is the address of **CCFG.BL_CONFIG**.

If a mechanism is made to set the memory map according to that flash, it would be nice.

g-oikonomou, 2016-10-13:
One definitely has to adjust the linker script to begin with. I think all CC1300s come as the same package, so I don't think pin mappings need to be adjusted.

lovelesh, 2016-10-13:
@g-oikonomou 
If we are sticking to 7x7mm chips then only memory map needs to be changed. If this works out, the same logic will work out with cc26xx series MCUs as well.

g-oikonomou, 2016-10-13:
We should adopt an approach whereby the linker script can be configured by the preprocessor. This is how the CC2538 code supports the chip's variants.

lovelesh, 2016-10-14:
@g-oikonomou if you have a plan then I can help you with this linker script.

wubai, 2016-10-19:
@lovelesh 
Your cc1310F64 can work now?  
I changed link-file cc26xx.ld ,but it's still useless.
If your devices can work, could you show me what to do?
Thank you!

lovelesh, 2016-10-19:
@wubai It still doesnot work. I am working on it.

cameronus, 2016-11-16:
@lovelesh Please let me know if there is progress on this. I really need to be able to flash my chip.

lovelesh, 2016-11-16:
@cameronus I am still not able to make the chips work. I will keep posting if I have any update.

wubai, 2016-11-30:
My cc1310F64 can work now!
First, we need to modify link-file contiki-master\cpu\cc26xx-cc13xx\cc26xx.ld.
Before:
```
MEMORY
{
    /* Flash Size 128 KB minus the CCA area below (88 bytes) */
    FLASH (RX) : ORIGIN = 0x00000000, **LENGTH = 0x0001FFA8**

    /*
     * Customer Configuration Area and Bootloader Backdoor configuration
     * in flash, up to 88 bytes
     */
    FLASH_CCFG (RX) : **ORIGIN = 0x0001FFA8**, LENGTH = 88

    /* RAM Size 20KB (PG2.1) */
    SRAM (RWX) : ORIGIN = 0x20000000, **LENGTH = 0x00005000**
	
	/* Application can use GPRAM region as RAM if cache is disabled in CCFG */
    GPRAM (RWX) : ORIGIN = 0x11000000, LENGTH = 0x00002000
}
```

After:
```
MEMORY
{
    /* Flash Size 128 KB minus the CCA area below (88 bytes) */
    FLASH (RX) : ORIGIN = 0x00000000, **LENGTH = 0x0000FFA8**

    /*
     * Customer Configuration Area and Bootloader Backdoor configuration
     * in flash, up to 88 bytes
     */
    FLASH_CCFG (RX) : **ORIGIN = 0x0000FFA8**, LENGTH = 88

    /* RAM Size 20KB (PG2.1) */
    SRAM (RWX) : ORIGIN = 0x20000000, **LENGTH = 0x00004000**
	
	/* Application can use GPRAM region as RAM if cache is disabled in CCFG */
    GPRAM (RWX) : ORIGIN = 0x11000000, LENGTH = 0x00002000
}
```
Second, we need to modify function ieee_addr_cpy_to(uint8_t *dst, uint8_t len).
It's  in the file contiki-master\cpu\cc26xx-cc13xx\ieee-addr.c .
we can not use IEEE_ADDR_LOCATION_SECONDARY. I don't know why, but the program stopped there,
when I use (uint8_t *)IEEE_ADDR_LOCATION_SECONDARY to get secondary address.
so, I modified it!

```
	for(i = 0; i < 8; i++) {
 #if 0
      if(((uint8_t *)IEEE_ADDR_LOCATION_SECONDARY)[i] != 0xFF) {
        /* A byte in the secondary location is not 0xFF. Use the secondary */
        location = (uint8_t *)IEEE_ADDR_LOCATION_SECONDARY;
        break;
      }
#endif
    }
```
Then,the cc1310F64 work well now!
I still can not find why (uint8_t *)IEEE_ADDR_LOCATION_SECONDARY makes cc1310F64 stop. 
If you know ,please tell me why! 
Thank you!
@g-oikonomou @lovelesh @cameronus

lovelesh, 2016-11-30:
@wubai GREAT WORK!!!
I will make the changes and try it.

> I wanted to discuss generic mechanism for all platforms. but cc13xx can be a good starting point.

arurke, 2016-12-01:
@wubai Did you have to make any compromises on the features you can include in contiki to make it fit on 64kB? (RPL, ipv6, etc.)

wubai, 2016-12-02:
@arurke I did not change or test other features like RPL,ipv6,tcp. I just use  it as a radio to transmit and receive message now.

arurke, 2016-12-04:
Ah ok. If I am not mistaken e.g. even the quite simple cc26xx demos are larger than 64kB by default. So would be interesting to learn about the efforts/any compromises needed to get a 6lowpan/rpl node running on 64k cc13xx.