contiki-os/contiki #1839: 2650 MCU waking every 50ms

jcole123, 2016-08-26:
After entering deep sleep it looks like the MCU is being woken up roughly every 50ms for some reason. Tested rpl-udp/very-sleepy-demo and on 2650STK as well as a custom 2650 board with no sensors on an oscilloscope and we're seeing the same narrow pulses every ~50ms on both boards. 

[image]

bkozak-scanimetrics, 2016-08-26:
I've noticed these regular power spikes during deep sleep as well.

I believe that they are most likely caused by the chip's 'recharge' functionality which is described (if only vaguely) in a number of the datasheets and user guides for the CC2650 (including the TRM). 

My guess is that this feature is meant to maintain some internal stored charge similar to how a DRAM refresh works but I've never been able to confirm this. 

I believe that you should fine that, even with the spikes, the average deep-sleep power consumption is still within the specifications found in the data-sheet.

casaroli, 2016-08-26:
Hello Billy,

Just an idea. What if we disable the DCDC in ccfg.c and see what is the
behaviour?

g-oikonomou, 2016-08-27:
This looks like the recharge to me as well, in which case it is _not_ a wake-up. What's the current peak of those spikes?

casaroli, 2016-08-27:
Hello,

I think that just for testing, it is easy to disable the DCDC converter in
ccfg and these recharges should go away.

At least we could rule out the software.

What do you think?

jcole123, 2016-08-29:
Disabled the DCDC converter in ccfg and we're still seeing spikes every ~50ms. It looks like it might be two separate events triggering every 100ms. The current peak on one is about 3mA while the other is <1mA.

jcole123, 2016-08-29:
So the 3mA spikes only started after a couple of days, here's a screenshot of the sub 1mA spikes we're seeing with DCDC off:
[image]

bkozak-scanimetrics, 2016-08-29:
> Just an idea. What if we disable the DCDC in ccfg.c and see what is the behaviour?

I think I had managed to stop the spikes once while debugging a separate issue. I don't remember exactly how but I think I modified some poorly documented registers that control the recharge algorithm. 

I'm not sure that there's any way to affect this by configuring the power supply (i.e. DCDC LDO, etc.). If you look at sections 6.3.1 and 6.3.2 of the TRM you'll see that you are also supposed to have a particular hardware configuration if you turn off the DCDC (and I believe that both sensortag and eval module are not in this configuration). 

> So the 3mA spikes only started after a couple of days

Could be related to the fact that the recharge function uses some "algorithm" and doesn't just turn on periodically. The 3mA spikes may only appear if you are spending enough time in sleep mode, for example. Again, however, I don't believe this is at all well documented so you may not have any way to verify that this is the exact expected behaviour without some more information from TI. 

If you can locate those registers I alluded to above maybe you can still run some experiments that can help to verify that this really is coming from the recharge function.

jcole123, 2016-09-08:
Is it possible to disable the DCDC programmatically or is changing the ccfg the only option? Even after manually turning the radio off we're still seeing spikes up to 3mA. Pink is DCDC_SW kicking in every 50ms. 
[image]

bkozak-scanimetrics, 2016-09-09:
I'm pretty sure that it's possible to disable the DCDC at any time.

Perhaps the ```PowerCtrlSourceSet``` function in the cc26xxware?

IIRC, some of the LPM routines end up modifying the power input mode so you may have to modify those if you want to be able to keep the DCDC off. 

Also, if you are experimenting with the LPM modes, you might want to try the [Contiki CC2650 LPM test project](https://github.com/bkozak-scanimetrics/CC2650_lpm_test) I made a while back (although I haven't used it in a while so maybe it doesn't work with latest contiki version). It was useful to me when I was studying issues with the sleep modes on the 2650.

jcole123, 2016-09-13:
So the 50ms spikes are recharge spikes, while the 3ma spikes are being caused by Contikimac. Not sure what Contikimac timer is causing the MCU to be woken up every 100ms, but with nullrdc and the radio off the spikes go from every 100ms to every second, which makes sense since the max sleep time is a second by default.

bkozak-scanimetrics, 2016-09-13:
It ought to be about 125ms between spikes with ContikiMAC in the default configuration. If you look in the default ```contiki-conf.h``` for the 2650 you'll find that ```NETSTACK_CONF_RDC_CHANNEL_CHECK_RATE == 8``` (i.e. 8Hz).

jcole123, 2016-09-13:
I have contikimac set to 2hz and the 3ma spikes still happen every 100ms even after turning the radio off.

adamdunkels, 2016-09-13:
With the default configuration, there is a periodic IPv6 timer firing every 100ms (https://github.com/contiki-os/contiki/blob/master/core/net/ipv6/uip-ds6.h#L168) maybe that's what you're seeing?

jcole123, 2016-09-13:
Yeah I noticed that and set it to fire every 20 seconds and it didn't make a difference.

jcole123, 2016-09-14:
It turns out I'm an idiot and it was the uip periodic timer causing the spikes. I must have changed the timer and flashed the wrong binary when I was testing. 

After changing UIP_DS6_PERIOD the only consistent spikes I'm seeing are the 2hz RX which is expected. Is there a reason why UIP_DS6_PERIOD is set so low? 
[image]

closed 2016-09-16 by jcole123