contiki-os/contiki #1840: can receive a packet from other Node even it's there packet in process

htm1112, 2016-08-26:
good evening every one,

I need to fined an answer for this question:  

is the Node can receive a packet from other Node even it's there packet in process?

thanks in advanced

g-oikonomou, 2016-08-27:
https://github.com/contiki-os/contiki/wiki#Help

closed 2016-08-27 by g-oikonomou

htm1112, 2016-08-27:
I don't know why you not very helpful, if I have knowledge, I will share it with others. I know your think this issue very simple or easy but for other it may be big step. 
if you give me the right answer will help me a lot. 
sorry for inconvenient and thanks.

alignan, 2016-08-27:
Posting an issue is not the right channel to ask for help, that is the purpose of the mailing list, it is not a matter of not be willing to help, but to avoid a misuse of the issue tracker.

htm1112, 2016-08-27:
OK. I'm sorry for this.