contiki-os/contiki #1843: how to use GPT timer A in cc1310 platform to drive etimer?

xioabudian, 2016-08-31:
Hi All,
I want to figure out the differences between using GPT timer A to drive etimer and using RTC to 
drive etimer.The I use the cc13xx driver library  files to write my program.I carefully refer to the 
technical reference manual,but it doesn't work.The interrupt is about 7.8125ms.I can't understand  why it doesn't work.Is there anything wrong with the codes.Can you give me some advices?
The following are codes:

void timer_isr(void){
  static int count=0;
  count++;

   if(count<256){
     ti_lib_gpio_pin_write(BOARD_LED,1); 
   }else{
     ti_lib_gpio_pin_write(BOARD_LED,0);
   }
   if(count>512)
    count=0;
}

void
clock_init(void)
{
  count = 0;

  bool interrupts_disabled;
  /* Disable and clear interrupts */
  interrupts_disabled = ti_lib_int_master_disable();

  if(ti_lib_prcm_power_domain_status(PRCM_DOMAIN_PERIPH) !=
     PRCM_DOMAIN_POWER_ON) {
    power_domain_on();
  }
 
  ti_lib_prcm_peripheral_run_enable(PRCM_PERIPH_TIMER0);
  ti_lib_prcm_load_set();
  while(!ti_lib_prcm_load_get());

  ti_lib_timer_configure(GPT0_BASE, TIMER_CFG_SPLIT_PAIR | TIMER_CFG_A_PERIODIC);

  ti_lib_timer_prescale_set(GPT0_BASE, TIMER_A, 90);

  ti_lib_timer_load_set(GPT0_BASE, TIMER_A, 2000);
  
  ti_lib_timer_int_register(GPT0_BASE, TIMER_A, timer_isr) ;

  ti_lib_timer_int_enable(TIMER_A, TIMER_TIMA_TIMEOUT);
  
   /* Re-enable interrupts */
  if(!interrupts_disabled) {
    ti_lib_int_master_enable();
  }
}

xukai871105, 2016-09-01:
I hope that you read more about etimer and rtimer in contiki. etimer and rtimer are easy to use. You can follow the example in contiki/examples/timers/all-timers.c or contiki/examples/cc2530dk/timer-test.c. These code can work well in CC1310 platform. Do not use contiki as 8051 MCU which you learned in the china college. 
建议你更换一种思路，其实学习contiki并不需要研究底层的硬件，contiki有很多组件，并不需要你从操作寄存器开始，从你的代码可以看出，etimer或者rtimer就可以满足你的应用。相信我，换种思路试试看。

g-oikonomou, 2017-01-10:
The main problem with GPTs is that they are in the PERIPH power domain, which we are shutting down when dropping to deep sleep. This means your GPT will no longer count while the chip is sleeping (the RTC keeps counting). Now you could force the GPT module to stay active during power-down, but this would cost a lot of energy so I personally consider it a bad design choice.

The second thing you need to consider is that you need something to wake the chip up and the RTC is a good choice here.

closed 2017-04-24 by g-oikonomou