contiki-os/contiki #1847: RPL nodes don't join new DODAG when BR replaced

AndrewBrannan, 2016-09-05:
I've got a network of ~20 custom CC1310 based devices running in a RPL mesh using 6LBR on a Raspberry Pi as the Border Router for the system.  The network runs great when the BR is started first and then all nodes are powered up afterward.

I am running into problems when I try to replace the original BR with a new one with the same RPL instance ID but with a new DODAG ID.  The new BR can see all the existing nodes as neighbours in it's neighbour table but none of them will be added to the new BR's routing table, and thus none of the nodes will be able to send data up to the internet.  If I turn on a new node, it will also never join the new DODAG, but will instead join the old DODAG which all the existing nodes had originally belonged to.  Even if I leave it for longer than `RPL_DEFAULT_LIFETIME` (which is set to 30 minutes), none of the old nodes will join with the new BR.

This situation could arise if there was a network running in the field and the network's BR failed.  If the BR had to be replaced with a new one with a new DODAG ID then all the nodes should switch to that new BR to continue sending data.  I think the problem is that the nodes aren't resetting their default root to use the new BR.

I'm using all the default values from rpl-conf.h.

Is this expected behaviour?  Are there any configuration settings that I'm missing that could influence this?  Thanks!

simonduq, 2016-09-05:
I think I have seen the same thing before, this is likely a bug. Thanks for reporting, we will investigate.

laurentderu, 2016-09-05:
See https://github.com/contiki-os/contiki/issues/1694 (still working on some of them though).

AndrewBrannan, 2016-09-05:
Ah @laurentderu , that looks great.  Any timeline on PRs for 1 or 2?

laurentderu, 2017-01-16:
Better late than never, see #1056 and #2057