contiki-os/contiki #1858: cc26xx-web-demo and very-sleepy-demo stop working after 600 seconds

magicm76, 2016-09-13:
Hello, 
I compiled examples (as they are) in contiki 3.x (DriverLib v0.46593) on different boards (SmartRF06 with cc2650em and SensorTag2.0) and: 
- cc26xx-demo is working well
- cc26xx-web-demo is rebooting after about 600 sec.
- very-sleepy-demo hangs after about 600 sec. 
I compiled example in Instant Contiki, with commands: 
- make TARGET=srf06-cc26xx name.bin CPU_FAMILY=cc26xx (for SmartRF06 with cc2650em)
- make TARGET=srf06-cc26xx BOARD=sensortag/cc2650 name.bin CPU_FAMILY=cc26xx (for SensorTAG)
Except rebooting and hanging applications seems working well. I looked for same problem in internet but I didn't find any solution

Thank You

schuschu, 2016-10-04:
I cannot tell you why it is happening but it I can observe the same behavior with a CC2650 running the web-demo and one as slip device (with 6LBR).

arurke, 2016-10-04:
Just for reference: I saw same issue reported on TI forums
http://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/p/544209/1991513

Seems they narrowed it down to COAP and/or MQTT app.

magicm76, 2016-10-04:
also the post in e2e.ti forum is mine, but is without answers

g-oikonomou, 2016-10-04:
Hi

I'll assume you are using latest version from master. Could you experiment with CoAP only and also making sure that the net-uart is disabled and report back?

P.S. (You don't really need to specify `CPU_FAMILY`)

schuschu, 2016-10-06:
Ok i tried on a launchapd to rule out some sensor process or similar: with udp, 6lbr, coap and mqtt it does also reset (as do my sensortags). when only enabling coap it does not occur. I'll try to do some more tests today.

magicm76, 2016-10-06:
Hi, 
I done more tests and I had a lot of problem but I found 1 configuration stable: 
Now, with master version of contiki, with SmartRF06 and cc2650EM as router with rpl-border-router, with motes as web-demo with only COAP active, with all sensor disabled in sensortag, system is stable. both SmartRF06 and sensortag motes seems stable. 
If I use sensortag web-demo as it is, with sensors active, motes reboot after near 600 seconds (SmartRF works well as it is in contiki example)
If I use MQTT instead COAP motes hangs or reboot, lot of time near 600 seconds
If I activate 6lbr-client  motes reboot after near 600 seconds
If I use as border router 6lbr over Beaglebone Black (as described in TI examples http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples) router hangs sometime, and after some time also motes. Application is very unstable.

g-oikonomou, 2016-10-06:
OK Thanks for the info, we need to look into this. I'm flagging it as a bug

g-oikonomou, 2016-10-14:
So this example either pushes stuff (6lbr client, net-uart, mqtt client) or it waits for another device to pull from it (http, coap).

If we disable the push logic, then the device will be sitting there doing pretty-much nothing. Are you people telling me it will crash even under those conditions?

The very sleepy demo seems affected by something reported in a different issue so let's leave it out of this discussion for now.

morphal, 2016-10-14:
I've made this config options in project-conf.h:
```
#define CC26XX_WEB_DEMO_CONF_MQTT_CLIENT      0 
#define CC26XX_WEB_DEMO_CONF_6LBR_CLIENT      0
#define CC26XX_WEB_DEMO_CONF_COAP_SERVER      1
#define CC26XX_WEB_DEMO_CONF_NET_UART         0
```
I've added this piece of code to the main loop of web-server:
```
if (ev == PROCESS_EVENT_TIMER && etimer_expired(&i_am_alive_timer)) {
    etimer_set(&i_am_alive_timer, CC26XX_I_AM_ALIVE_PERIOD);
    printf("[%lu]i am alive\n", clock_seconds());
}
```
Result:
'i am alive' messages stop appearing after 539 seconds.
It still crashes even with all push logic disabled

g-oikonomou, 2016-10-18:
@morphal Are you using sensortags / launchpads / something else?

morphal, 2016-10-18:
@g-oikonomou, 
6lbr + cc2531 usb dongle as slip-radio
sensortags running web-demo as 6lowpan nodes

magicm76, 2016-10-18:
About push question in my tests COAP and net-uart functions are stable, MQTT and 6lbr client make system rebooting. 
I used in my tests both SmartRF06 with cc2650EM and Sensortag.

CogPy, 2016-11-25:
I can confirm to see a similar thing with a modified very-sleepy-demo.

If I configure the MPU9250 to work in interrupt mode, I have to reconfigure it every 10'. I can live with it but it's an ugly workaround.

larskholte, 2016-12-29:
I am having the timeout occur in cc26xx-web-demo even with the bare minimum configured options.
```
/* Enable/Disable Components of this Demo */
#define CC26XX_WEB_DEMO_CONF_MQTT_CLIENT      0
#define CC26XX_WEB_DEMO_CONF_6LBR_CLIENT      0
#define CC26XX_WEB_DEMO_CONF_COAP_SERVER      0
#define CC26XX_WEB_DEMO_CONF_NET_UART         0
```
I also commented out all calls to `init_sensor_readings` and `init_sensors`. No luck.

I inserted a piece of code into the thread main loop, as @morphal did, to print 'i am alive'. These messages continue long after the node stops responding to pings and serving webpages.

[image]

I'm using CC2650 Rev 1.3 SensorTags, a CC2531EMK slip radio, and a BBB. The timeout seems to occur in cc26xx-demo as well as cc26xx-web-demo.

OSPro, 2017-01-10:
Same issue with cc2650 sensortag here after testing many cases (Can not ping sensortag after about 10 minutes). I am using BBB as 6lbr router and cc2650 launchpad for slip-radio function.  Trying with a pc with ubuntu os instead of deabian on BBB. Any one solved this issue? 
Thanks.

g-oikonomou, 2017-01-10:
Some additional info in #2006

OSPro, 2017-01-11:
Tested with an ubuntu pc instead of one debian os on BBB and same issue happening.  This issue almost same to this link on 6lbr issues: https://github.com/cetic/6lbr/issues/231
But i see one note thing: when i repower one sensor node, all another sensor nodes will also come back (can ping them again). I tested with 2 cc2650 sensortags. So i guest this is 6lbr server issue and need to keep connection to 6lbr server by sending data to it periodically, maybe from another coap client service.
Thanks.

laurentderu, 2017-01-12:
To rule out one potential issue, could someone having the problem retest with the following compilation flag enabled :

   #define UIP_CONF_ND6_SEND_NA   1

The ~600s looks strangely similar to the default NUD period...

dbjorklu, 2017-01-12:
Setting  ```#define UIP_CONF_ND6_SEND_NA 1``` seems to work for me.
But as I wrote in cetic/6lbr#231 my nodes are not resetting, they are running and I keep receiving push messages from them, just can't reach them through the 6lbr border router after 10 mins.

OSPro, 2017-01-13:
@laurentderu u are correct. My sensortags working fine after setting "#define UIP_CONF_ND6_SEND_NA 1". I can get sensor values using coap after 10 minutes now. Thanks.
@dbjorklu can you say more about your issue? i do not see that issue when use  "#define CC26XX_WEB_DEMO_CONF_6LBR_CLIENT  1". I only have issue with COAP stop working after 10 minutes and can not ping sensor nodes then. Thanks.

g-oikonomou, 2017-01-15:
@laurentderu do we have a somewhat more generic problem with NDP that does not limit itself to this particular platform? I've not followed the most recent discussions

dbjorklu, 2017-01-15:
I do see the same problem with an AVR platform (and actually I am not running exactly the cc26xx demo on cc2650, furthermore I have some custom stuff in 6lbr).
I'm going to test with rpl-border router instead of 6lbr if I get the time.

g-oikonomou, 2017-01-15:
That could help a lot

laurentderu, 2017-01-16:
@g-oikonomou There is a problem when UIP_ND6_SEND_NA is set to 0, it means the host will not respond to incoming NS at all, and so another host relying on NDP will assume the peer is not reachable and, if RPL is activated, the corresponding route is removed. I created #2073  to tackle this issue.
On the other hand, the rebooting problems reported are not related.