contiki-os/contiki #1862: Problem with using the 20 bit addressing with MSP430 on Cooja

huy-truong, 2016-09-15:
Hello there,
I am trying to using the Relic-toolkit ( a cryptography library) for Wismote running on Cooja.
The compiling  is failed because of the shortage of memory. Then I use an option as the below command in order to force MSP430 use the 20 bit addressing space.
$make hello-world.wismote TARGET=wismote TARGET_MEMORY_MODEL=large
The compilation goes well, but  when I started the mote on Cooja, it run for a while and then it stops and raising the error.

darkopetrovic, 2016-11-29:
I've a similar issue while trying to compile a project on the wismote platform:

```
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.2/../../../../msp430/bin/ld: coap-example.wismote section `.data' will not fit in region `rom'
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.2/../../../../msp430/bin/ld: section .vectors loaded at [0000ff80,0000ffff] overlaps section .data loaded at [0000fcfa,0000ff81]
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.2/../../../../msp430/bin/ld: region `rom' overflowed by 2 bytes
```

The same way, if I use `TARGET_MEMORY_MODEL=large` the compilation goes well but the simulation doesn't work:

```
[java] MSP430: *** IOUnit write to non-existent IO at $0000
     [java] Stack Trace: number of calls: 4 PC: $17664
     [java]   uip_ds6_periodic (mac-sequence.c) called from PC: $19b24 (elapsed: 1287)
     [java]   process_thread_tcpip_process (local in tcpip.c) called from PC: $14c46 (elapsed: 1363)
     [java]   call_process (local in process.c) called from PC: $14e20 (elapsed: 1400)
     [java]   process_run (mac-sequence.c) called from PC: $05f34 (elapsed: 2240)
     [java]  WARN [Thread-19] (MspMote.java:229) - 6: # MSP430[VOID_IO_WRITE]: *** IOUnit write to non-existent IO at $0000
```

I've installed gcc from `http://simonduq.github.io/resources/mspgcc-4.7.2-compiled.tar.bz2`. 

I've tried equally to specify the linker script file with `LDFLAGS += -Wl,-T msp430.x` and by copying this latter in the root of the project along with the memory.x and periph.x files but no success.

ArcticXWolf, 2017-04-06:
@darkopetrovic Did you find any solution to this? I am getting the exact same errors, when compiling Contiki with tinyDTLS.

darkopetrovic, 2017-04-07:
Hi,

No I'm sorry. I didn't found and understand why this problem appends. I think this is due to the fact that MSP430X is experimental in mspgcc 4.7.2. 

My code has slightly changed from the last time and hopefully this error didn't happens anymore with the actual libraries I've included in the project.

You can still try with the `TARGET_MEMORY_MODEL=large` option or with another platform like Z1 and see if it works.

ArcticXWolf, 2017-04-10:
Ah, too bad. But thanks for answering!

I simply do not get, why cooja gets these stacktraces when compiled for msp430x :/

fabricio-godoi, 2017-04-10:
Hey guys,
I was messing with some others codes to compile and run under Cooja environment, and I got a lot of these errors. Try to check if all drivers are properly setup, check for possibles semantic errors (null pointers, unsigned variables, etc). This error is caused when the Program Counter (PC) register of MSP is set to zero (IOUnit write to non-existent IO at $0000), making the all code get lost. So check if the function uip_ds6_periodic is properly setup, or even the mac-sequence.c.

Hope you guys figure this out!

Best regards!