contiki-os/contiki #1872: Strange packets on mDNS request

aignacio, 2016-09-22:
Hello,
I'm using the example rpl-border-router in cooja with localhost webserver  (tunslip6 -a 127.0.0.1 aaaa::1/64) and the hello world in the node, and then when I use the **resolv_set_hostname(...**  to change the hostname of devices, the node just not answer anymore to icmpv6 or another request. The avahi daemon is running on my linux machine and in the wireshark the first packet sent by node is something unknow....
[image]

**TARGET**=z1

Anybody had the same problem?

vsaw, 2016-09-23:
Wireshark shows a ICMP error message that the destination Port of the initial UDP packet is 0. This is an invalid port number as ports must be > 0. Are you sure your initial packet is correct?

aignacio, 2016-09-23:
Since I opened the tuneled port (tun0) wireshark was waiting for packets and more than one times this thing happen...Other thing noted is that the node just answer to ping req. one time and die after when I active resolve_set_hostname