contiki-os/contiki #1878: CC1310 + 6lbr crash under heavy traffic

emreerisgen, 2016-09-30:
Hi,

I have 20 cc1310 nodes(nullrdc,csma) and 1 bbb runnning 6lbr with cc1310 slip radio connected via UART. I am facing a problem where RX part of the radio of the slip radio stops working and that problem occurs almost random. I am not sure it is just slip-radio or nodes too. I have a proven scenario on my setup that reproduce the issue so i will describe it.

Start to GW, start the nodes everything goes ok.
Power of the GW, wait for more than 10 minutes, Restart the GW none of the nodes visible. Sometimes some of the nodes comes to the 6lbr sensors page, in neighbour table and route table but they cant be reachable. 

I am not sure it is CC1310 specific problem or problem is caused by slip radio but i do know that other people saw that problem too. You can follow this link [http://lists.cetic.be/archives/6lbr-dev/2016-September/000777.html](url) Raising the ACK wait time helps the problem a little bit but when the number of nodes gets bigger it occurs again. 

I am adding my logs too if anyone want to look at them. [https://www.dropbox.com/s/dpqd9c4maeqq1fy/Case1.zip?dl=0](url)

tldr: CC1310 nodes or slip radio stop working under heavy traffic.

AndrewBrannan, 2016-10-03:
I've had the exact same issue with some custom CC1310 hardware, haven't found the root cause yet.  As you mentioned, the problem gets bad once there are >20 nodes on the network.

arurke, 2016-10-03:
Hi, are you seeing the problem appear randomly or only at 6lbr reboot?

I have not tested large scale, and as such have only experienced problems at 6lbr reboot, which is the original issue I posted in the mailing list. To sum it up with a 6lbr+cc1310 SLIP and one cc1310 node programmed to transmit every second, all settings (ContikiMAC, ACK timings etc.) default:

1. Node and GW is up, node transmits every sec to GW. Everything works fine.
2. GW is rebooted. The node will transmit a lot because of the 1 sec interval and retransmission (since it does not receive ACK)
3. GW is up again. SLIP radio acts strangely: Does not transmit ACK (so node correctly keeps TXing) and does not forward received frames to 6lbr. However, it transmits DIOs (so it is not completely dead + the band is evidently not completely saturated).
4. Rebooting node has no effect. Only way to fix is to turn off node, and then reboot GW.

If I increase the interval at the node, the fault will happen more seldom. My conclusion was that something goes wrong in the cc1310 if it boots up in an rf environment with a lot of traffic. Laurent at 6lbr does not believe it is caused by 6lbr, so that leaves the cc1310 as likely culprit. (i.e. this could probably be reproduced with two nodes and no GW as well (if it is not tied to SLIP SW))

laurentderu, 2016-10-03:
From what I observed, the issue seems to occur only when the BR restarts. The network looks fine before, but once the BR and the slip-radio have restarted the problem pops up.

From the pcap file and the nullrdc and contikimac traces, it seems that the slip-radio and the nodes no longer transmit Acks (in the pcap you see many times the same message without a ack, even late). Without Ack every messages are therefore retransmitted by CSMA.

I see a few root causes :
* the radio code (rf-core) somehow get stuck in an unexpected state and so stop reporting incoming messages to the higher layer.
* CSMA retransmission is too fast and so collide with the ack messages (but there are virtually no corrupted messages in the pcap, unless sensniff on CC13xx does not log corrupted messages)
* The radio does CCA before sending the ACK, and due to the amount of traffic, the Ack is never send
* something else :)

arurke, 2016-10-03:
I doubt point 2 & 3 since I saw DIOs being transmitted from 6lbr via SLIP as
normal (as far as I could tell)

Andreas

>

sgoveia, 2016-10-14:
@g-oikonomou,
I'm very interested in this one, as I hope to scale the CC1310's up to ~200+ nodes. Thank you so much for all the great work!

g-oikonomou, 2016-10-14:
Hi guys,

First thing we should note here: ACKs are _not_ sent by the hardware, they are created and sent by s/w.

The CC1310 lets NullRDC / ContikiMAC create and send those ACKs. With some other radios I think it's done by the RF driver itself, but they are still s/w controlled.

> the radio code (rf-core) somehow get stuck in an unexpected state and so stop reporting incoming messages to the higher layer.

Possibly. I am thinking maybe it receives something before it's put itself in a state ready-enough to correctly RX it (hence no ACK).

> CSMA retransmission is too fast and so collide with the ack messages (but there are virtually no corrupted messages in the pcap, unless sensniff on CC13xx does not log corrupted messages)

I think I'm with @arurke here since the SLIP radio is sending out DIOs.

To answer @laurentderu , If a sensniff receives a frame with a FCS error, it will not log it. This is something I don't quite like, but this is how the current Contiki example works.

> something else :)

* What if it's NullRDC that's stuck in a funny state and refuses to send ACKs?
* Has anyone tried without Crypto? (just in case....)

g-oikonomou, 2016-10-14:
I meant to also ask: Has anyone tried this with a border router instead of 6lbr+slip-radio?

g-oikonomou, 2016-10-17:
This is where we enable the global interrupt flag (` ti_lib_int_master_enable();`) in the startup sequence:

https://github.com/contiki-os/contiki/blob/master/platform/srf06-cc26xx/contiki-main.c#L171

As a sanity check, could someone try moving this line to immediately before the main `while()` loop (currently L235)?

g-oikonomou, 2016-10-17:
@arurke Andreas, am I right in interpreting your post that you can have seen this with a single mesh node? What is your node sending out every second?

arurke, 2016-10-17:
@g-oikonomou Yes. The node is transmitting a COAP (using the coap library) frame every sec with a destination addr outside the network. I have a sniffer pcap file of the scenario I explained above if interested.

With regards to your question on border router instead of 6lbr: No, not tested.

I can try to reproduce this (some weeks/months since I first reported this on 6lbr mailing-list). Can also try your suggestion with the flag.

g-oikonomou, 2016-10-17:
Thanks Andreas

Are they CoAP Observation or are you pushing data with Put/Post?

I want this resolved as a matter of prio, so any attempts/reports on the various thoughts I mentioned above would help. For example, are you using crypto?

arurke, 2016-10-17:
COAP post/put. No crypto. I think the major hint his no-ACKs but still TX DIO - I am guessing there are just a few ways this could happen (we double-checked the L2 addresses are correct). I will see if I can reproduce and test your suggestion tonight.

g-oikonomou, 2016-10-17:
Also, resetting the mesh node does not fix things, does it?

arurke, 2016-10-17:
Correct

arurke, 2016-10-17:
Tried your suggestion, but no difference.

I will add one disclaimer; for some reason the node stops retransmitting after 15-20 seconds of trying. (as I can remember from earlier it would transmit indefinitely. Maybe there has been changes to max rank or similar(?). I did not bother to debug it now). Anyway, this meant I could not reboot the GW since it takes more than 20 sec to reboot it. What I did instead was to power off/on the SLIP alone - @laurentderu: should 6lbr tolerate this?
I was able to recover the situation doing this so I believe it is fine (see step 5 below)

Test was:
1. One node tx COAP PUT every sec. Everything works fine
2. Power off SLIP, node will retransmit a lot
3. Power on SLIP, node still retransmit, no ACKs, GW tx DIO periodically
4. Reboot node. When it receives a DIO from GW, will start transmit/retransmit of COAP (no ACKs received)
5. Power off node. Power off/on SLIP.  Power on node. Everything works.

Bleeding-edge contiki as of today, ipv6/SLIP example with QUEUEBUF_CONF_NUM and BUFFER_SIZE commented out,  460800 baud. 6lbr 1.4.0.

g-oikonomou, 2016-10-17:
Got you, thanks, that is very helpful. I shall try to reproduce. I'd first try the interrupt thing if I'm honest.

laurentderu, 2016-10-18:
@arurke power cycling the Slip-Radio is not an issue for the BR. The reset will trigger a restart of the BR as the slip device becomes unavailable and it will wait until it's available again.

arurke, 2016-10-18:
@laurentderu I am certain the 6lbr log @ 10 did not show any restart of the BR, should it? (I unpowered the SLIP only for a few seconds btw).

@g-oikonomou my changes
```
--- a/platform/srf06-cc26xx/contiki-main.c
+++ b/platform/srf06-cc26xx/contiki-main.c
@@ -168,8 +168,6 @@ main(void)
 
   fade(LEDS_RED);
 
-  ti_lib_int_master_enable();
-
   soc_rtc_init();
   clock_init();
   rtimer_init();
@@ -233,6 +231,8 @@ main(void)
 
   fade(LEDS_ORANGE);
 
+  ti_lib_int_master_enable();
+
   while(1) {
     uint8_t r;
```

laurentderu, 2016-10-18:
If it's a USB-based slip-radio you should have seen something in the log. On the other hand, a with a vanilla UART slip-radio you should indeed see nothing as the host device is still configured, but it should not cause troubles AFAIK.

arurke, 2016-10-18:
Ok. It was UART and it seemed to work fine.

g-oikonomou, 2016-10-18:
Andreas yes, that's the change I was suggesting.

UART device powercycle should not show in the log, unless some USB/UART chip is in use (most cases) and this also gets powercycled. If the `/dev/whatever` disappears, I bet you'll see something from 6lbr.

g-oikonomou, 2016-10-22:
I can confirm this. I'm flagging it as a bug.

g-oikonomou, 2016-10-23:
Folks, can you try with this branch?

https://github.com/g-oikonomou/contiki/tree/bugfix/cc13xx/no-acks

@emreerisgen @AndrewBrannan @arurke

arurke, 2016-10-23:
a night well spent @g-oikonomou  :)
I employed it to my SLIP radio only and it works like a charm.

FYI I also checked out ebd1d and it also worked.

g-oikonomou, 2016-10-24:
Thanks Andreas

I think I have a theory here (a better one now than on Saturday anyway!)

Within `prop-mode.c`, I expect a device in the broken state to report the following:

`transmitting()` returns 0
`rf_is_on()` returns 0
`rf_core_is_accessible()` returns 0
`smartrf_settings_cmd_prop_rx_adv.status` has value of 0

Can users confirm this please

arurke, 2016-10-24:
At which point do you want to check those values? I never got debugging to work properly so would need to put it in code somewhere.

g-oikonomou, 2016-10-24:
I often just bind a printout to a button press

arurke, 2016-10-24:
Using a EM card ad-hoc attached to beaglebone UART. I added a printout to the 3sec etimer in slip_radio_process. When in bad state, printout came very irregular and the match your expectations:
```
transmitting(): 0, rf_is_on(): 0, rf_core_is_accessible(): 0
smartrf_settings_cmd_prop_rx_adv.status: 0
```
for reference, good state:
```
transmitting(): 0, rf_is_on(): 1, rf_core_is_accessible(): 1
smartrf_settings_cmd_prop_rx_adv.status: 2
```

g-oikonomou, 2016-10-24:
Yep, exactly what I was hoping to see

g-oikonomou, 2016-10-24:
So, `transmit()` checks the state of the radio at the start. If it is off, we assume this is because of ContikiMAC power cycling. So we turn on, TX, turn back off.

What happens here is that, due to high traffic, at some point we receive more frames than we handle (presumably at startup). Because of that, all RX buffers fill up and then `PROP_CMD_RX_ADV` exits with code 0x3801 `PROP_ERROR_RXBUF` ("No available Rx buffer at the start of a packet"). Next time we try to TX, RX is no longer running, so we TX and then turn the radio back off.... Ergo, the radio ends up powered down except when about to TX.

Normal power cycling with ContikiMAC will restore the radio back to a sane state at next cycle. However, because slip-radios and border-routers force the radio to be always "on", we end up stuck in this hung state.

The proper workaround is to correctly handle `PROP_CMD_RX_ADV` exit causes when caused by an error.

AndrewBrannan, 2016-10-25:
Thanks for looking into that @g-oikonomou.

I tried out your [branch](https://github.com/g-oikonomou/contiki/tree/bugfix/cc13xx/no-acks) and saw no change from contiki master unfortunately.  Though, looking through that branch it seems like it doesn't address the `PROP_ERROR_RXBUF` error you mentioned above?

My test set-up consists of 6lbr 1.4.0 running on a RPi with a custom cc1310 device as a slip radio and 10 custom cc1310 sensor nodes each doing a CoAP GET transaction every 10 seconds, very similar to @arurke's setup.  It will often start up fine but within 10 minutes the slip radio will lock up as above.  I'll check the returns of `transmitting()` etc tomorrow.  Is there anything else you'd like to see the status of in a failed state?

g-oikonomou, 2016-10-25:
Thanks and you are right, we're not handling the RXBUF error condition, we are just trying to make sure it doesn't happen at startup. As I say, proper fix required.

After some internal discussions, I am now starting to think whether it may be related to UART speed and also perhaps whether you are using debug statements?

AndrewBrannan, 2016-10-25:
I've tried it with UART speeds of 115200, 230400 and 460800 and found it didn't make a difference. Ditto on the debug statements,  all disabled even the initial Contiki version message.

g-oikonomou, 2016-10-25:
That's very helpful Andrew, thanks

AndrewBrannan, 2016-10-27:
I can confirm @arurke's observations. 

When the slip radio is in a good state I see:
```
transmitting() = 0 
rf_is_on() = 1 
rf_core_is_accessible() = 1 
smartrf_settings_cmd_prop_rx_adv.status = 2
```

g-oikonomou, 2016-10-27:
Andrew yes, this is exactly the first error state that triggers the problem. Before `.status` goes to 0, it will first go to 0x3801.

When `CMD_PROP_RX_ADV` ends with 0x3801, `RFC_DBELL_O_RFCPEIFG.RFC_DBELL_RFCPEIFG_RX_BUF_FULL` should get asserted. This flag won't stay high, we will clear it as part of `on()`. So if you were to add the value of the flag in your test script, it should be high when status is 0x3801.

I'm thinking about ways of handling this:

* Better intelligence when checking whether the radio was off
* Acknowledge this interrupt and handle the condition
* Increase RX buffers to 4 (which is what we do in the IEEE mode driver)

AndrewBrannan, 2016-10-27:
Makes sense.

Where is the logic to check the state of the radio before `transmit()` at the moment?

g-oikonomou, 2016-10-27:
It's in `rf_is_on()` (L243), called by `transmit()` at L641 as well as at other places.

Basically if the power domain is on _and_ RX is running, we are in RX. Otherwise we assume the radio was off and we therefore assume ContikiMAC and turn it back off. What happens here is that the power domain is on, but RX is _not_ running, which causes the logic to fail.

AndrewBrannan, 2016-10-27:
Yep, gotcha.

Seems like it should do 2 & 3 for sure, gracefully handle the interrupt in `cc26xx_rf_cpe0_isr()` (I think?) and increase the number of rx buffers.  

As for adding more logic to `transmit()`, should a call to transmit really be concerned about whether the there's an error in on the rx side of things?  It feels like transmit should simply stop rx if it's on, do it's thing and then restore whatever state it found the radio in (as it does now), leaving handling rx issues to the rx interrupt handler.

g-oikonomou, 2016-10-27:
I agree with your statement about transmit. It should definitely fire up the radio if required and also stop RX (as it already does). As you say, it should definitely not handle the error state, but it does have to correctly restore state, which at the moment it fails to do IMHO.

We can map any interrupt source to either vector. I'd probably go with CPE1 for this one.

g-oikonomou, 2016-12-10:
Second attempt at fixing this in this branch:

https://github.com/g-oikonomou/contiki/tree/bugfix/cc13xx/no-acks-2

I'm wondering whether simply increase the number of RX buffers will make this go away.

arurke, 2016-12-10:
Unfortunately, it did not

g-oikonomou, 2016-12-10:
OK thanks for trying, this is now starting to confuse me quite a bit. I cannot immediately imagine the circumstances under which we can receive 4 frames faster than we can process them!

arurke, 2016-12-11:
Not sure if helpful, but I increased it to 8 - and it works reliably

g-oikonomou, 2016-12-11:
!

BaN-Co, 2016-12-14:
Yes, it works fine now, after I increased the number of the buffers.
Before this fix, in the send_packet() function of the contikimac.c file, during the NETSTACK_RADIO.channel_clear() loop before sending the packet, I've received maximum of 2 interrupts on receive in the cc26xx_rf_cpe0_isr() and in some cases after this the problem occured.
After I increased the number of buffers to 8, during the NETSTACK_RADIO.channel_clear() loop I sometimes receive 3 interrupts on receive in the the cc26xx_rf_cpe0_isr().
So it was the root of the problem.
I think we need to make the number of buffers user defined, because in some users cases the traffic will be low and there is no need to high number of buffers, but in other cases there will be high traffic and the number of buffers needs to be increased.
Along with increasing the number of buffers we need to handle this problem correctly. Maybe we should count the number of used buffers: increment in the interrupts on receive cc26xx_rf_cpe0_isr(), decrement when we handle the packet. Then once all buffers will be used up, we can turn off the receiver, if it was on, and turn it back on, when the buffers will be freed, if need to.

g-oikonomou, 2016-12-15:
In the same branch, I pushed a first take of how I would imagine the number of RX buffers to be easily configurable from conf files (It was high time we changed this particular code anyway). Please test if you have time.

g-oikonomou/contiki@b03ef66c4b7511e88bf8f3b46bc260635f101002

I'd still like to understand the deeper cause for this, but increasing buffer count will have to do for now.

BaN-Co, 2016-12-15:
@g-oikonomou Thank you, I'll try it out.
As I understand this, it seems like the the PROP_ERROR_RXBUF error is causing the RF core domain to turn off.
And I don't think that just increasing the buffer count will be enough, as someone may have the unpredictable case in which there will be VERY high traffic and even 4 buffers won't be enough, so I think we need to correctly handle the case, when the PROP_ERROR_RXBUF will appear and turn off the RF core domain. To turn the RF Core back on in the send_packet() function, or to count the number of free buffers in the interrupts and disable receiver to avoid this error occuring altogether.

g-oikonomou, 2016-12-15:
Yea this was just a stub, should be increased to whatever. I agree this is likely to reappear under different conditions.

As I have already said, the `PROP_ERROR_RXBUF` should be handled somehow, but powecycling the radio is not a good way to do this. When the upper layer wants the radio to stay on, the driver should respect that.

BaN-Co, 2016-12-15:
@g-oikonomou Can you help me with understanding the contikimac send_packet() expected behavior?
Precisely in this code:
```
    /* Check if there are any transmissions by others. */
    /* TODO: why does this give collisions before sending with the mc1322x? */
  if(is_receiver_awake == 0) {
    int i;
    for(i = 0; i < CCA_COUNT_MAX_TX; ++i) {
      PRINTF("m");
      t0 = RTIMER_NOW();
      on();
#if CCA_CHECK_TIME > 0
      while(RTIMER_CLOCK_LT(RTIMER_NOW(), t0 + CCA_CHECK_TIME)) { }
#endif
      PRINTF("0x%04X%d|", smartrf_settings_cmd_prop_rx_adv.status, ((rfc_propRxOutput_t* )smartrf_settings_cmd_prop_rx_adv.pOutput)->nRxBufFull);
      PRINTF("n");
      if(NETSTACK_RADIO.channel_clear() == 0) {
      PRINTF("o");
        collisions++;
        off();
        break;
      }
      PRINTF("p");
      off();
      t0 = RTIMER_NOW();
      while(RTIMER_CLOCK_LT(RTIMER_NOW(), t0 + CCA_SLEEP_TIME)) { }
      PRINTF("q");
      PRINTF("0x%04X%d|", smartrf_settings_cmd_prop_rx_adv.status, ((rfc_propRxOutput_t* )smartrf_settings_cmd_prop_rx_adv.pOutput)->nRxBufFull);
    }
  }
```
We need to check, whatever there is any communication going on in the channel, and if there is any, then we need to skip sending the packet, right?
In the loop we check the RSSI for a few short intervals and if there is no communication (RSSI < threshold), then we continue to send the packet.
And we **ASSUME**, that we won't receive any packets in the meantime, while we check the RSSI.
Then what happens, if those RSSI checks didn't happen while the packets were transmitted, but in betwen them and in this RSSI loop we received a few packets (up to 3 for my tests) and continue to send the packet?
It'll first send the packet and then it'll wait for ACK and when it'll check, whatever the ACK was received or not, it'll check **THE FIRST** received packet, as we **ASSUMED**, that there shouldn't be any other received packets in the queue, because before RSSI loop check we checked, whatever there is any packets pending with NETSTACK_RADIO.pending_packet(), so because first packet in the queue wasn't ACK, it'll be seen as collision and noack for the sent packet.
If I got it right, then how we should handle this?
- Disable receiver while checking RSSI, even with the contikimac_keep_radio_on = 1?
- Shorten the RSSI check intervals, when the contikimac_keep_radio_on = 1?
- Check the NETSTACK_RADIO.pending_packet() after the RSSI check, if the contikimac_keep_radio_on = 1?
I think the best will be to shorten the RSSI check intervals and also to add the NETSTACK_RADIO.pending_packet() check in the RSSI loop check, if the contikimac_keep_radio_on = 1.

g-oikonomou, 2016-12-16:
I think this particular issue isn't really the best place to discuss ContikiMAC's logic or propose potential fixes to it (of which I think it needs many).

I don't like the idea of disabling the receiver while checking RSSI, because this would disrupt ongoing reception.

We are not exactly assuming we won't get any packets while checking RSSI. If we are in the process of reception, sampling RSSI should signal this. I do not understand how you say that you have observed reception of up to three packets while transmitting?

I'm looking for a sequence of events that make things bug manifest itself, and I think most of the frames that fill RX buffers are strobes of the same frame. You are probably trying to tell me something here, but I am not understanding it right at this moment I'm afraid.

g-oikonomou, 2016-12-16:
Also, forgot to ask, did you get a chance to check this piece of code with more generic / configurable RX buffer count?

BaN-Co, 2016-12-16:
@g-oikonomou
> Also, forgot to ask, did you get a chance to check this piece of code with more generic / configurable RX buffer count?
Yes, it works fine.

Regarding the ContikiMAC issue, it's related to this issue, I'll try to describe it a bit more detailed:
I'm talking not just about checking RSSI in general, but about how the checking if there are any transmissions by others is handled in the send_packet() function, before the packet is sent (calls of the NETSTACK_RADIO.channel_clear() in the loop with delays).
In the send_packet() function there are 3 main parts:
- Check that the channel is clear with the loop RSSI check (using NETSTACK_RADIO.channel_clear() call) before sending the packet.
- Send the packet.
- Wait for the ACK.
The issue is occuring only if the contikimac_keep_radio_on == 1.
When the send_packet() is called from the qsend_packet() and qsend_list(), it's assumed inside them, that the receiver isn't awake and the is_receiver_awake parameter is passed with the 0 value to the send_packet() function, even if the is_receiver_awake should be 1, because the contikimac_keep_radio_on is 1.
So, when the send_packet() is called with the parameter is_receiver_awake == 1, even though the receiver is on by the contikimac_keep_radio_on == 1, before it send the packet it'll first enter the RSSI check loop for some (quite big) time.
And this is the problem, we check if the channel is clear with the receiver on for quite the time, during which we can receive a lot of packets, without being able to process them.

g-oikonomou, 2016-12-16:
> Yes, it works fine.

Thanks, great,

>  In the send_packet() function there are 3 main parts:

OK yep. I'm following now.

So we have two nodes: a slip-radio + 6lbr and a mesh node A. `contikimac_keep_radio_on` is set to 1 on the slip-radio, which is where the problem is occurring.

The `is_receiver_awake` variable is with reference to the other node (i.e. not the one running the code, but the one intended to receive). So in our example, if the TXing node is node A, then `is_receiver_awake` will be with reference to the slip-radio, and the other way round. Unfortunately, there is no way to predict that the other device will be always on. Not with the current code base anyway.

> So, when the send_packet() is called with the parameter is_receiver_awake == 0, even though the receiver is on by the contikimac_keep_radio_on == 1, before it send the packet it'll first enter the RSSI check loop for some (quite big, ~34ms in my case) time.

Yes getting stuck for 34ms sounds like a little bit of a problem indeed! But can you clarify whether this wait happens on the slip-radio side or on the node A side? I am assuming that you're talking for the slip-radio device, right?

This look does checks the channel N (default 6) times and there are two waits involved: `CCA_CHECK_TIME` and `CCA_SLEEP_TIME`. With our configuration, the former is ~625usec and the latter is about 5ms... In our `contiki-conf.h`:

```
#define CONTIKIMAC_CONF_CCA_CHECK_TIME            (RTIMER_ARCH_SECOND / 1600)
#define CONTIKIMAC_CONF_CCA_SLEEP_TIME            (RTIMER_ARCH_SECOND / 210)
```

I'm sure I must have been thinking about something when I set those to these values, but can you sanity check here? The default for `CCA_SLEEP_TIME` is `RTIMER_ARCH_SECOND / 2000`, which is about 10 times smaller. What happens if we set it back to this value?

BaN-Co, 2016-12-16:
>So we have two nodes: a slip-radio + 6lbr and a mesh node A. contikimac_keep_radio_on is set to 1 on the slip-radio, which is where the problem is occurring.

In my tests, the problem 
>The is_receiver_awake variable is with reference to the other node (i.e. not the one running the code, but the one intended to receive).

Ok, I've got it now, I've just misunderstood it.
>But can you clarify whether this wait happens on the slip-radio side or on the node A side? 

This wait happens on the slip-radio side.
In my tests, the situation, when the problem occures, is as follows:
Network: 6lbr - slip-radio <-> NodeA <-> NodeB
The problem occures, when the NodeA and NodeB communicate between themselves and the slip-radio start to send packet to one of the nodes in between their communication and the clear channel checks happens in between the nodes communication, but the slip-radio still receive a few packets in between the rssi checks.

I don't really understand the need in the CCA_SLEEP_TIME delay, is it the delay, that should let us with high chance to check the channel clearance during the consecutive communication in the air not to check all N times in between the packets if the CCA_SLEEP_TIME is too big and not to check all N times in between the packets, if the CCA_SLEEP_TIME is too small? Maybe the CCA_SLEEP_TIME value is increased, because the data rate 10 times smaller, comparing to the other CPUs, when that code was written?

>What happens if we set it back to this value?

I'll try to increase this value and test it, but this will only have an influence on the normal nodes with contikimac_keep_radio_on == 0.
I think, that we need to add in the clear channel check loop the NETSTACK_RADIO.pending_packet() check, and if during the channel check the packet is received, then we return from the send_packet() due to the collisions before sending with the MAC_TX_COLLISION.

BaN-Co, 2016-12-16:
So as all timing is right, I'll repeat my point:
I think, that we need to add in the clear channel check loop the NETSTACK_RADIO.pending_packet() check, and if during the channel check the packet is received, then we return from the send_packet() due to the collisions before sending with the MAC_TX_COLLISION.

g-oikonomou, 2016-12-16:
>  Also, where did the CCA_CHECK_TIME value for the cc26xx-cc13xx came from? Is it measured value? I couldn't find anything in the datasheets.

Pure trial and error :)

g-oikonomou, 2017-01-15:
#2071 with should fix this till we can configure improved timings. The slip radio will need built with a higher number of buffers, eg:

```
#define PROP_MODE_CONF_RX_BUF_CNT        8
```

closed 2017-01-21 by g-oikonomou

tussock-cam, 2017-02-22:
Just want to note that I still get this issue, even with 8 rx buffers and only 4 nodes connected. Seems to happen when they have a poor connection to the BR, and is generally the nodes themselves that freeze. I've been testing with nullrdc so contikimac doesn't force them back into rx mode. As a quick workaround I check for `IRQ_RX_BUF_FULL` in `rf_core.c` and unconditionally call `NETSTACK_RADIO.on()` to "fix" it.

arurke, 2017-02-22:
Interesting, can you explain in more detail how you reproduce?

[12 more replies]

PR #2071 (2017-01-21): Make the number of Prop Mode RX Buffers easy to configure
This pull makes the number of RX buffers for CC13xx prop mode easily configurable via a simple define.

Can be used to fix #1878, possibly also #1933 and #1999.