contiki-os/contiki #1881: Serial communication  issue on zolertia remote-revb

elvisvogli, 2016-10-02:
Hello, 

I have just received some  new revision B zolerti Re-motes but I am having some issues on making them work. At first I thought it was a bootloader problem because I couldn't flash them. The following error happened most of the times. 

> 
> arm-none-eabi-objcopy -O binary --gap-fill 0xff hello-world.elf hello-world.bin
> Flashing /dev/ttyUSB0
> Opening port /dev/ttyUSB0, baud 500000
> Reading data from hello-world.bin
> Firmware file: Raw Binary
> Connecting to target...
> ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
> 

However if I insisted several times  (on average one successful flash on 10 tries), I succeed to program the motes with  the hello-world application. After connecting the mote the hello-world application is not printing any data in the terminal. I have to press many times the reset button to get printed some times a partial output or very rarely the full output of the hello-world.  Therefore I believe that there is an issue related to the serial communication and not only to the bootloader. 

Do you have any  idea why is this happening? am I missing anything ?

In the following the output of the  hello-world application (at least for the cases when there was some output because most of the times nothing was received). 

> Contiki-3.x-2904-g61 0:12:4b:00:06:0d:b2:  
> 
> Contiki-3.x-2904-g61
> 
> Contiki-3.x-2904-g61
> 
> Contiki-3.x-2904-g61
> 0:12:4b:00:06:0d:b2:4a
> Net: sicslowpan
> 
> 0:12:4b:00:06:0d:b2:
> 
> Contiki-3.x-2904-g61
> 
> Contiki-3.x-2904-g61
> 0:12:4b:00:06:0d:b2:4a
> Net: sicslowpan
> 
> Contiki-3.x-2904-g61
> 
> Contiki-3.x-2904-g61
> 
> Contiki-3.x-2904-g61
> 0:12:4b:00:06:0d:b2:4a
> Net: sicslowpan
> 
> 0:12:4b:00:06:0d:b2:
> 
> Contiki-3.x-2904-g61
> 0:12:4b:00:06:0d:b2:4a
> Net: sicslowpan
> 
> Contiki-3.x-2904-g61
> 0:12:4b:00:06:0d:b2:
> 
>

alignan, 2016-10-03:
What commit are you using? Could you please paste the `git log` output? Also please go to `tools/cc2538-bsl` and check the commit number

elvisvogli, 2016-10-03:
The last commits are as follow::

```
commit 6157dce0b55dd337cd8c2e4050f3000b2a982985
Merge: fd881be 1758eaa
Author: Antonio Lignan <antonio.lignan@gmail.com>
Date:   Wed Sep 28 23:05:17 2016 +0200
    Merge pull request #1877 from alignan/pull/fix-zoul-doc
    Zoul: updated pin-out images
```

The bsl  last commit is : 

```
commit c6100a7794c7b530923145c03e37412013a4551e
Author: George Oikonomou <george@contiki-os.org>
Date:   Tue Jun 21 00:00:58 2016 +0100
    Update version number to 2.1
    Changes:
    * Fixed (hopefully for good) ACK/NACK issue with Zolertia Zoul (RE-Mote & Firefly)
    * Added workaround for CC2538 chips that have their IEEE address written on InfoPage 
    * Added the ability to invert the DTR/RTS lines when performing the sequence to invok
    * Added support for more CC2538 variants (PG1.0, F64KB, 8KB SRAM)
```

alignan, 2016-10-03:
If you do as shown in:

https://www.youtube.com/watch?v=6vg4f7U_kF4

Does flashing works?

elvisvogli, 2016-10-03:
Sorry for the delay but I just got the hands on motes. The problem however is still present even with the Bootloader unlocked.

alignan, 2016-10-03:
Does this happen with all the RE-Motes, or just this one? it is strange, as with the bootloader unlock sequence it should work

elvisvogli, 2016-10-03:
I confirm that It is happening  with all the motes which is vary strange.
Also I am testing the zoul-demo application now. It does not print any data in the serial but when i send a string ending in \\r it seems that is responding because it toggles the leds all the times. It seems that there is an issue only in the direction from the mote to the PC.

mbenalaya, 2016-10-04:
Hello, 

I have the same issue with the same motes.
Any solution ?

Thank you in advance

alignan, 2016-10-04:
Mmm probably the best would be to check each case individually, and then update the topic with the findings, please send an email to support@zolertia.com, include the serial number of the devices or the order number.  Thanks for reporting!

closed 2016-12-04 by alignan

alignan, 2016-12-20:
https://github.com/contiki-os/contiki/pull/2008