contiki-os/contiki #1888: CC2650 Sensortag web demo freezing

morphal, 2016-10-13:
Hello,

I've successfully set up 6lbr + cc2650 sensortags running contiki web demo.
Everything works great except I have similar problem to [discussed here](https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/t/544209)

Sensortag 'freezes' after 539 seconds of perfect operating.
The differences to linked discussion are:
- freeze period is 539 seconds (stable reproduces)
- usually sensortag just 'freezes' but once i saw it rebooted instead of 'freezing'

I can definitely assert, that:
1. web server process doesn't receive any event before device freezing (i was expecting it can receive PROCESS_EVENT_EXIT/PROCESS_EVENT_SERVICE_REMOVED)
2. contiki cc26xx-demo works perfect without this issue
3. period of 539 seconds is stable
4. freezing does not depend on active modules in project-conf.h. It freezes with all combinations of COAP server, MQTT and UDP options enabled/disabled

My troubles a even worse since I can debug only via serial port messages
Can anyone please help me to handle the reason?

source code and bin attaches:
[freezing-demo-bin.zip](https://github.com/contiki-os/contiki/files/527262/freezing-demo-bin.zip)
[freezing_demo.zip](https://github.com/contiki-os/contiki/files/527263/freezing_demo.zip)
web-demo folder: mincontiki\applications\cc26xx\cc26xx-web-demo\

Thank you!

g-oikonomou, 2016-10-13:
Related to #1858 perhaps?

morphal, 2016-10-13:
yes, it is really seems to be related to #1858
I think we can close this issue and i will join investigation there
thanks!

closed 2016-10-13 by morphal