contiki-os/contiki #1891: JN516x micromac radio driver is too slow

laurentderu, 2016-10-18:
I tried to make a JN5169 talks with a CC2650 using ContikiMAC and NullRDC, without success. The issues are twofold :
The packet acking on the JN516 is done by software and is too slow for the CC2650 with default parameters. Using NullRDC, I saw that the ack packet comes between 45 and 50 ticks. The default timeout (RTIMER_ARCH_SECOND/2500) is 26 ticks.
On the other hand, when the JN516 sends a packet to the CC2650, the ack is never received as the JN516 radio has not switched to Receive mode yet when the ack packet is sent by the CC2650. Forcing SW Ack sending on the CC2650 slows them enough to be received by the JN516X.

Can someone with more knowledge about the JN516 confirm or infirm the issue ?
I believe that switching to MAC mode instead of PHY mode should solve the issue as, if I understood the JN doc correctly, the Ack waiting and sending is done by the hardware. But I would like to be sure before implementing it :)

simonduq, 2016-10-18:
Right, the radio drivers for JN516x were primarily written for TSCH, with most H/W support disabled, and we indeed only support S/W acks. Switching to MAC mode is the way to do it yes. Contributions highly welcome :)

Are you talking about JN516x ticks or CC2650? Note that on the JN, the default config runs rtimer at 16MHz.

laurentderu, 2016-10-18:
The CC2650 ticks indeed, I was sloppy in my description of the issue :)
MAC mode it will be then, my only worry is if it will be able to support LLSEC. The mac frame has no support for the security field and security attributes, so they should be included in the payload buffer.

simonduq, 2016-10-18:
Right, the H/W in MAC mode doesn't seem to parse security headers into tsMacFrame, and will likely end up in the payload buffer. Maybe not straightforward to parse with the existing framer functions (as you won't have a readily-available buffer that contains the header).

For LLSEC, note that `jn516x-ccm-star.c` implements provides drivers for the H/W coprocessor.

neophob, 2016-12-12:
@simonduq do you have a concrete todo list for the JN516x, what could be improved?

laurentderu, 2016-12-12:
I will create a PR with the HW MAC mode (without security) this week for review.

neophob, 2016-12-12:
@laurentderu great, happy to test.

simonduq, 2016-12-13:
@neophob there is a very minimal todo list in https://github.com/contiki-os/contiki/tree/master/platform/jn516x

Feel free to PR with more suggestions :)

neophob, 2016-12-20:
@laurentderu any news?

laurentderu, 2016-12-22:
@neophob I should know better and never gives unrealistic deadlines :) but it's available now, see https://github.com/contiki-os/contiki/pull/2012

neophob, 2016-12-22:
@laurentderu unrealistic deadlines? tell me one dev out in the world that does estimate too low, haha. thanks, will check it out!